Ultimaker/Cura #5617: [4.0.0] Tree supports should be smoothed

calculuschild, 2019-04-12:
**Application Version**
Cura 4.0.0

**Platform**
Windows 10 x64

**Printer**
Folgertech Prusa i3 
[PI3_Orc Soldier 1.3mf.zip](https://github.com/Ultimaker/Cura/files/3073192/PI3_Orc.Soldier.1.3mf.zip)

**Steps to Reproduce**
Slice a file using Tree Supports. I used these Settings:
[image]
[image]

**Actual Results**
"I've noticed the outer edges have very fine detail, often under a line width, and the printer following these makes all sorts of racket. I have some aggressive accel settings, etc, but the rest of the print doesn't do this (small zig-zags being the exception).

 As you can see, there is a lot of detail in there which is so small that it hardly makes a difference that it is there﻿."

**Expected results**
"The segment in question (9 lines) could easily be reduced to 2 without changing anything about the support it applies. The uppermost layer/layers may well have stringent accuracy requirements, but the bulk of the tower could be smoothed to within 1-2 line widths without hurting the print. This would reduce file size, wear on printer, print times, etc etc."

**Additional Information**
Resulting tree supports are extremely jagged and rough:
[image]

The text and image from this issue were borrowed from a post on the Ultimaker forums that never got any replies, "[https://community.ultimaker.com/topic/21444-suggestion-tree-supports-need-smoothing/](https://community.ultimaker.com/topic/21444-suggestion-tree-supports-need-smoothing/)". This wasn't me, but since I have run into the same issue I am saving myself some typing.

calculuschild, 2019-04-12:
As part of this, when looking at the TreeSupport.cpp file, I see that the maximum allowed displacement is 1/4 of a line:
`support_layer.simplify(circle_side_length * (1 + diameter_angle_scale_factor_this_layer), line_width >> 2); //Deviate at most a quarter of a line so that the lines still stack properly.
`
However, in the polygon.simplify function this is intended to be the allowed distance **squared**. Am I misinterpreting that, or should the allowed displacement in TreeSupport be `(line_width >> 2 * line_width >> 2)`?

In any case, how much of an issue would it really be if the line was shifted by maybe a half or full line_width? Would the layers above and below not also be simplified in a similar manner such that they stack properly anyway?

Ghostkeeper, 2019-04-12:
> However, in the polygon.simplify function this is intended to be the allowed distance squared. Am I misinterpreting that, or should the allowed displacement in TreeSupport be `(line_width >> 2 * line_width >> 2)`?

There are actually two `.simplify` functions: One for `Polygon` (a simple polygon) and one for `Polygons` (a complex polygon). The one for simple polygons expects squared parameters. The one for complex polygons doesn't. This one calls the one for complex polygons, so it looks like the current code is correct in this regard.

The jagged edges arise from tree support placing circles of support down (the "branches") and then moving the circles closer together. The circles then merge as they collide, which causes these corners. Aside from the low resolution of the circles themselves to keep it somewhat performant, that is. A smoothing step is in fact applied to this, as you've found.

The limits of 1/4 line width have been experimentally found using my printer at work. If you put this too high you'll find that it puts lines next to each other and this kills the strength of support which is critical for tree support. Indeed your small line could be filtered out though. I guess it should, but it's hard to find a way to make it work for every geometrical case.

calculuschild, 2019-04-12:
Thanks for the reply!

> There are actually two .simplify functions: One for Polygon (a simple polygon) and one for Polygons (a complex polygon).

Where is the complex version? I am having trouble finding it and am just curious about its code.

Perhaps for supports like this where preserving details is not as important as on the actual model itself, there could be some check for minimal area between three points and if it falls below a threshold, remove the middle point? i.e., three points along the contour form a triangle, and if it is a very small triangle below some fraction of the current polygon area, or the current layer circle size (like the little zig-zags above), or even a global value, straighten it out. For instance, I can't picture a need for details like that if they are smaller than the smallest circles that the tree will generate.

> If you put this too high you'll find that it puts lines next to each other and this kills the strength of support which is critical for tree support.

I'm not sure what you mean by "putting lines next to each other". Do you mean subsequent layers have lines offset from one-another so they are printed nearly in mid-air? My intuition is that simply running the simplification a second time, screening for only very small segments but allowing larger displacement would be enough to get those little nubs out without shifting the larger main lines too much.

Then, I think it could be possible to remove some of the zigzags by screening for short segments between two long segments that are almost parallel, like this:

[image]

Ghostkeeper, 2019-04-15:
> Perhaps for supports like this where preserving details is not as important as on the actual model itself, there could be some check for minimal area between three points and if it falls below a [...]

That's basically how it gets at the maximum displacement already:

https://github.com/Ultimaker/CuraEngine/blob/5d8d07e339b7ec1ff72929898d1dbe54367d0264/src/utils/polygon.cpp#L320

In the call to the Simplify function, this maximum displacement is set to be a quarter of the line width. This is then (inversely) divided by the length of the shortcut to arrive at the height of that triangle which is the displacement we're comparing with.

It also has a limit on how far it simplifies in the line length though in order to not simplify the original circles if they are small. Maybe that limit should be removed.

calculuschild, 2019-04-15:
Right, I can see the steps it goes through. As you say, it does calculate the area of the triangle, but that is used only as a means to compare against the height of the triangle as the limiting dimension. In the model itself, this preserves details since sharp angles are determined by that triangle height. I.e., narrow but tall triangles will be preserved, even if the area is very small. Only relatively flat triangles are cut:

[image]

However in the support structure, preserving those narrow, tall triangles is undesirable. So I should rephrase: perhaps the triangles should be evaluated on the area **only**, regardless of height vs length. Or check both height and length and if *either one* is very short, cut the vertex, regardless of the size of the other. In particular, we want to focus on removing *concave* features since they originate from the gaps between circles. That may be a way to avoid damaging the small circles.

[image]

So, here's my proposed algorithm so far:

1. Search vertices along the path as per the normal `simplify`.
2. If the removed segment is shorter than current layer circle edge
3. and the removed triangle is below some minimum specified area (regardless of triangle height)
4. and the removed area is concave
5. remove

Ghostkeeper, 2019-04-16:
That sounds like it could work.

calculuschild, 2019-04-16:
Awesome. I'm glad the Cura team is open to thoughts. I wouldn't expect this to be worked on right away but I'll subscribe to this issue and hopefully hear about some good news down the road!

Ghostkeeper, 2019-04-17:
We decided to not make a specialised simplification algorithm for this part because it's too much work to maintain. However we also figured out that the testing done for this particular detail was not very extensive so we're not too sure of it.

So we decided to use the current `simplify` function but make the deviation a bit bigger. I made it use the Tree Support Collision Resolution setting for that. An explanation is here: https://github.com/Ultimaker/CuraEngine/commit/5c70524f28aeb092d069419b27869e1e46d389c0

Increasing the Collision Resolution to 0.22mm or so smooths out the support fine for me and reduces printing time of support by 1.5% or so. Nothing appears to go wrong when looking at layer view.

closed 2019-04-17 by Ghostkeeper