Ultimaker/Cura #5621: Layer Preview Stoping

RocketNut, 2019-04-12:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

RocketNut, 2019-04-12:
The layering preview is stopping at random layers. Even when it is not even at the top layer. Like at 75 or 105 when there is over 1000 layers!

Ghostkeeper, 2019-04-15:
What do you mean with "stopping"? Does it pause the replay when you press play? Or maybe it doesn't generate the layer view completely?

For me the layer view seems to be complete. The correct number of layers in total, anyway. My 1cm cube has 99 layers (initial layer height 0.27, layer height 0.1).

What are your reproduce steps, actual results and expected results? It'd help if you filled out the template more completely.

RocketNut, 2019-06-15:
I just upgraded to 4.1 and the same thing happening. I started a model at layer 1 then to bed and when I check on it in the morning it was still on layer 1. I had to pause it and then resumed it. It then did layer 2 and again it stopped when that was done. 

Here is a link to the STL file I am trying to preview.

[https://www.dropbox.com/s/fr5pfvtq87nkuyw/TurboRockEngine%201.5meter%20Fuel%20Feed%20Dome%20Defuser%20Disk%20D2%20SP%2025mm.stl?dl=0](url)

When I did this model with version 3.4 with out any problems. Layer perview works as acpected, no stopping.

Ghostkeeper, 2019-06-17:
I'm unable to reproduce it so far. I've been watching it go through a whole print without stopping.

Maybe it only happens when the computer goes into sleep mode? That could be system dependent.

I wouldn't know why you'd leave the preview running over night though... It's not like it's going to store any results for you and it's pretty terrible as a screen saver!

RocketNut, 2019-06-18:
It's strange using version 3,4,1 without any problems!!!!

I have sleep mode turned off, because I am running some eng simulation that also takes a long time to complete.

I ran it overnight to see if I just not letting it to complete a layer.

I guess it just my machine. Go ahead close this ticket. Thank you for your help you deserve 5 stars,

Ghostkeeper, 2019-06-18:
All right. If someone else finds this too, maybe we can find the common denominator between the two computers then. I'll close this as requested until then.

closed 2019-06-18 by Ghostkeeper