Ultimaker/Cura #5622: weirdness with the display of Compensate Wall Overlaps

yet-another-average-joe, 2019-04-12:
**Application Version**
<4.0>

**Platform**
Windows 10

**Printer**
Tevo Tornado

**Steps to Reproduce**

Play with :

- Compensate Wall Overlaps
- Compensate Outer Wall Overlaps
- Compensate Inner Overlaps

**Actual Results**

- unchecking "Compensate Wall Overlaps" does not hide the 2 subitems (outer and inner), but uncheck them
- uncheck "Compensate Wall Overlaps", then check the subitems : "Compensate Wall Overlaps" remains unchecked ;  "Compensate Wall Overlaps" becomes unresponsive (probably disabled ?), and gets some sort of hilight : black with a white surrounding rounded square . It has probably the state BST_INDETERMINATE | BST_UNCHECKED
- check "Compensate Wall Overlaps" ; then uncheck the 2 subitems ; "Compensate Wall Overlaps" gets checked, grayed and disabled (BST_INDETERMINATE | BST_CHECKED in Win32 API vocabulary, this means that the control has the BS_3STATE or BS_AUTO3STATE attribute ; QT issue ?)

**Expected results**

Unchecking Compensate Wall Overlaps should hide its 2 subitems

**Additional Information**

[image]

undeterminate state, unchecked and disabled :

[image]

undeterminate state, checked and disabled :

[image]

Synching controls state with object attributes can make a coder crazy...

Ghostkeeper, 2019-04-15:
This seems to behave as I'd expect, but different from what you expect.

Compensate Inner/Outer Wall Overlaps are subsettings of Compensate Wall Overlaps in the same way that Wall Speed is a subsetting of Print Speed.

If you modify the Print Speed but leave the Wall Speed untouched, the Wall Speed will adjust automatically based on the ratio of the new Print Speed vs. the original Print Speed. By default, the Wall speed is exactly half of the Print Speed so if the Print Speed is modified it'll modify itself to stay at half the Print Speed. In the same way, Compensate Outer/Inner Wall Overlaps take the value of Compensate Wall Overlaps unless overridden. This is why unchecking Compensate Wall Overlaps automatically also unchecks Compensate Outer/Inner Wall Overlaps.

However, you can still override the subsettings. You could change the Wall Speed to a manual value. At this point, the Wall Speed is no longer half the Print Speed and will not change along any more. A little fx icon is shown to signal that it used to change along with something but is no longer doing that. If you were to change the Infill Speed, Wall Speed, Top/Bottom Speed, Support Speed and Prime Tower Speed, then you've changed all subsettings of the Print Speed setting and changing the Print Speed won't have any effect on the print any more, so we disable it. In your second and third cases, you've overwritten both the Compensate Outer Wall Overlaps and the Compensate Inner Wall Overlaps settings so the parent Compensate Wall Overlaps setting has no effect any more, and Cura disables it. This probably results in that `BST_INDETERMINATE` flag that you are seeing.

Maybe the actual problem is that due to the colour scheme it's not clear that the parent Compensate Wall Overlaps is disabled?

yet-another-average-joe, 2019-04-15:
Ooops ! I forgot to complete the issue report ! It was about Minimum Wall Flow...

I was describing the GUI behaviour of the 3 checkboxes, and focused on their interactions (probably because I love coding GUIs). I was somewhat confused, and then forgot to add that whatever "Compensate Wall Overlaps" and its subitems are set to, Minimum Wall Flow edit field stays diplayed and enabled (not grayed, not hidden). 

In another issue, you wrote that Minimum Wall Flow should disappear from the interface if "Outer and Inner Compensate Wall Overlapse" are both unchecked :

> Do you mean that Minimum Wall Flow should be disabled if Compensate Wall Overlapse is set to false ?
> 
> Your answer: Yeah the setting should disappear from your interface.

and :

> Yeah. I'm not able to reproduce that though. In the setting definition it says that the setting is visible if > either Compensate Outer Wall Overlaps is enabled or Compensate Inner Wall Overlaps is enabled.
> Maybe some weirdness with the display of the checkmarks. We've had that before.

Not really understanding these 3 checkboxes interactions, I described them, and forgot the real object of the issue.

And thanks for the explanations ! There are some parameters I don't really understand. I stick with the english interface because most of the informations on the web are in english, but it is not always easy for advandced features ! There was an amazing help system plugin, and it disappeared with 4.0 :(((

Ghostkeeper, 2019-04-16:
I looked into it and found some weirdness that prevented the setting from being hidden when both checkmarks were disabled in the per-extruder settings. Should be fixed for 4.1.

commit 7b62eb95 (2019-04-16): Make minimum wall flow also settable per extruder

It was settable per mesh but not per extruder, but we want it to be settable per extruder as well, causing all meshes printed with that extruder to use that setting value. That's how most of these settings work (like Outer Wall Speed and such).
There are some exceptions to this where a setting only makes sense if applied in per-object settings, such as Infill Mesh or support blockers, but this is not such a setting.

Fixes #5622.
closed 2019-04-16 by Ghostkeeper