Ultimaker/Cura #5623: Cura treats filament as 3mm even when set to 1.75 - impossible to use custom filament.

asdasd-123, 2019-04-12:
**Application Version**
4.0.0

**Platform**
Windows 10

**Printer**
Custom Build

**Steps to Reproduce**
- Machine setting set to 1.75mm diameter filament.
- Custom filament set to 1.75mm diameter
- At 1.75mm diameter, file should take 0.55m of filament.
- Cura slices to 0.21m (as if it were 3m)
- To force it to slice for the correct diameter you need to change the machine filament settings to 3mm then BACK to 1.75mm. But this also resets the material selected.
- Select the custom material again, and cura reslices the object at 3mm filament again. 
- Start from step 1 and get stuck in a loop for 20 minutes until you give up and post here.

I've included a full recording below in the gif. 
Please help :)

[image]

pcollingwood, 2019-04-14:
I too have this problem, but was just going to uninstall and reinstall Cura. I will try what you have done and see if it's still misbehaving.

asdasd-123, 2019-04-14:
I found that "duplicating" one of the default filaments and changing their values doesn't have this issue, so I've had to re-make every custom filament now using that method.

pcollingwood, 2019-04-14:
I'll try that. For me I just can't get it to keep the 1.75mm setting on one of my Delta's. On the other I'm still using 3.6.0 because I've got a MKS touch screen and there's no plugin for that in 4.0 yet. Aaaaargh!

Ghostkeeper, 2019-04-15:
Unable to reproduce this so far. I don't have your model so I must get at my own numbers here, so this is what I did.

1. Add Custom FDM Printer. Printer is still at 2.85mm filament.
2. Load 1cm cube and let it slice. The result is 0.11m of filament used.
3. Go to Machine Settings and change the compatible material diameter to 1.75mm. Material is automatically switched from PLA to PLA_175 (but they have the same name so it's not visible in the interface other than that the material diameter is 1.75mm).
4. Let it slice again. The result is 0.30m of filament used, so that is the result of using 1.75mm filament.
5. Add a custom material (not duplicating) and switch to it. The slice result still says 0.30m filament. So it seems to use 1.75mm filament.

Duplicating from PLA at step 5 has the same effect.

asdasd-123, 2019-04-15:
Thanks, 
I'm happy to send any config files or models needed. 
l can only get custom filaments working by duplicating existing filaments.

diegopradogesto, 2019-05-03:
If you created your own definition file for your printer, be sure that you also create a definition file for the extruder. Take for instance the 3Dator printer definitions as an example:

- Printer definition: https://github.com/Ultimaker/Cura/blob/master/resources/definitions/3dator.def.json
- Extruder definition: https://github.com/Ultimaker/Cura/blob/master/resources/extruders/3dator_extruder_0.def.json

asdasd-123, 2019-05-03:
Hi, I created neither files manually, but I did create both inside Cura

Ghostkeeper, 2019-05-06:
What are the files that you have been using then?

asdasd-123, 2019-05-08:
I've just installed a fresh Cura and used the steps in the original post to arrive at the issue. As above, I have not setup and am not using any specific files. 
If the process of creating the profiles inside Cura has created any files then you need to tell me which files you want copies of.

Many thanks

diegopradogesto, 2019-05-09:
I also can't reproduce this. I see the same behaviour as @Ghostkeeper , that is expected.

closed 2020-06-20 by no-response[bot]