Ultimaker/Cura #5626: Handling filenames with a dot generates the wrong filename

JesConsuegra, 2019-04-13:
**Application Version**
 4.0.0

**Platform**
 Win10 64bit
[PTS2.hombro_casita_opuesto_C4_ABSHI.3mf.zip](https://github.com/Ultimaker/Cura/files/3076047/PTS2.hombro_casita_opuesto_C4_ABSHI.3mf.zip)

**Printer**
 PRusa i3 MK2

**Steps to Reproduce**
 just save a gcode with a filename with an early dot in the filename.

**Actual Results**
 the output gcode file gets truncated after the first dot. i.e. PTS2.hombro_casita_opuesto_C4_ABSHI.gcode gets truncated to PTS2.gcode when saved.

There is an issue closed for 3.2.1: [3.2.1] File saving: filenames cut by first dot #3609 opened on Apr 1, 2018 by acorderob 
If you want to save the gcode after slicing, the filename gets truncated after the first dot.

**Expected results**
 keep the filename intact

**Additional Information**

Ghostkeeper, 2019-04-15:
Doesn't reproduce for me on Linux, so maybe only on Windows again.

Liger0, 2019-04-15:
Not really. I can't reproduce it on windows either.

rburema, 2019-04-16:
I can't reproduce this either (on Windows) with both 4.0.0 and the current master.

JesConsuegra, 2019-04-24:
I still have the problem. At any time I put a dot in the filename, it gets cut after slicing.
[image]
Which information do you need, (besides the one already sent)?
Thanks.

Ghostkeeper, 2019-04-25:
Maybe a log file would help. Otherwise we might just need to wait if someone else also experiences this and find out what patterns there are in what the two people have in common to see what's causing this.

JesConsuegra, 2019-04-25:
Ok, I'll try to find out how to generate a log.
What it is installed is the standard downloadable installer.
When you say it cannot be reproduced with the current master, does it mean that there is a source  updated version, different from the downloadable installer?
I'll try to test it on Linux, I got a machine with Ubuntu Mate 18.

JesConsuegra, 2019-04-25:
Might it be related to localization?
I have both win10 and Cura 4.0.0 in english but some locals in spanish (i.e. Euro currency, dd/Mm/yy date and spanish keyboard, Etc.).

JesConsuegra, 2019-04-25:
Ok, I'll try to find out how to generate a log.
What it is installed is the standard downloadable installer.
When you say it cannot be reproduced with the current master, does it mean that there is a source  updated version, different from the downloadable installer?
Also I use a SSD hard drive as OS only and all my files are on external USB devices.

I'll try to test it on Linux, I got a machine with Ubuntu Mate 18.

JesConsuegra, 2019-04-25:
Also I use an OS only SSD drive and all my files and data are on external USB drives.

Ghostkeeper, 2019-04-25:
Here are some instructions on where to find the log: https://github.com/Ultimaker/Cura#logging-issues

Yeah rburema and I tested this on current master (which is the prototype version for Cura 4.1), and rburema and Liger0 tested this using the stable version 4.0.0 as well that you were using.

I don't think that this issue has anything to do with the type of drive you're using, unless the file name contains special characters that may not be supported by the file system of that drive, but practically any modern file system supports unicode nowadays so it shouldn't be a problem. Localisation, maybe, but then it will be specific to Windows and I can't test that.

JesConsuegra, 2019-04-25:
I remember why I am not using it on Linux: my machine is 32bit and there is no 32bit Cura anymore :(
Anyway I look for the log.

JesConsuegra, 2019-04-25:
I've just switched Cura from english to spanish and the problem has disappeared...
Back to english and the problem is back again.
Pretty sure it has to be related to localization...
Enclosed cura.log from my machine.

[cura.log](https://github.com/Ultimaker/Cura/files/3117585/cura.log)

JesConsuegra, 2019-04-26:
At the next time I started the laptop (with the spanish locals and Cura 4.0.0 in spanish) the problem does appear again...

Ghostkeeper, 2019-04-29:
Hmm, the log seems to mention the correct file name:

```
2019-04-25 17:04:20,157 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [126]: Writing to [E:/_3Ddissenys/Casita_juguete_Marion/Casita_2019/TC4PG_2.hombro-casita-d.gcode]...
2019-04-25 17:04:20,169 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [144]: Writing to Local File E:/_3Ddissenys/Casita_juguete_Marion/Casita_2019/TC4PG_2.hombro-casita-d.gcode in text mode
```
Or maybe that happened to be the case when it was working correctly?

closed 2020-06-20 by no-response[bot]