Ultimaker/Cura #5627: First I used the BETA 4.0.0 then about 2 weeks ago start to struggle to get it open, now I removed it and INSTALL Ultimaker Cura 4.0.0 and still it was truckling to open, I play with Win 10 compatible problem solving and it did not open then all the sudden it open I do not now if it will open if I close it again ?

dwaalmj, 2019-04-13:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

Ghostkeeper, 2019-04-15:
I don't know either. Could you try? I mean, keeping it open indefinitely is not a viable solution either, is it.

Perhaps your virus scanner could be blocking it from opening.

Vandrasc, 2019-06-27:
Hi @dwaalmj, as Ghostkeeper mentioned above, Cura could be blocked by your virus scanner. Unfortunately, we cannot reproduce the issue described by you, but could you please retry with our 4.1 Cura version and check your virus scanner? Please reopen the ticket if indeed still an issue. 

Thank you!

closed 2019-06-27 by Vandrasc