Ultimaker/Cura #5628: Cura crashes on launch, only sows "User Agreeent"

NovaKeks, 2019-04-14:
**Application Version**
 3.6.0

**Platform**
 Windows 10

**Printer**
 could not select

**Steps to Reproduce**
 opening cura

**Actual Results**
 cura crashes after showing a blank windows titled User Agreement

**Expected results**
 Cura should start

**Additional Information**
 [cura.log](https://github.com/Ultimaker/Cura/files/3077556/cura.log)

Ghostkeeper, 2019-04-15:
Does it still happen in Cura 4.0?

What if you accept the user agreement?

MichaelPinpin, 2019-04-16:
I also have the same problem. Can u guys take a look at my log as well ?

thx
[stderr.log](https://github.com/Ultimaker/Cura/files/3083105/stderr.log)

NovaKeks, 2019-04-16:
@Ghostkeeper

I haven't tried Cura 4.0 yet. I didnt realize its already out!

And I didnt get the chance to accept the  user agreement because the programm crashes before the agreement itsself is even shown.

diegopradogesto, 2019-06-20:
Hi @MichaelPinpin , your version of Cura seems to be the one that Lulzbolt created. Although it could be related, we can't control what happens in that software.

For you @NovaKeks , could you please take a look in the latest Cura 4.1 to see if it works?

closed 2020-06-19 by no-response[bot]