Ultimaker/Cura #5632: [master] project save dialog shows incorrect material

smartavionics, 2019-04-15:
**Application Version**

master

**Platform**

Linux

**Printer**

Kosssel XL

**Steps to Reproduce**

Load previous project, change some setting, go to save project, look at dialog.

**Actual Results**

Dialog shows a material associated with extruder 1 that I have never used.

**Expected results**

Material shown for extruder 1 should match the currently selected material.

**Additional Information**

[image]

Ghostkeeper, 2019-04-15:
Unable to reproduce this. I don't have the Kossel XL definition file though. Here were my full reproduce steps:

1. Add Kossel Pro as closest approximation.
2. Load a cube.
3. Save the project to a file. Extruder 1 material shows PLA.
4. Load the project file again.
5. Change the printing speed to 59 (down from 60).
6. Save the project file again.

In the dialogue it still shows PLA for me.

smartavionics, 2019-04-15:
OK, but where does that value come from? Why is it not influenced by the material selected for extruder 1?

smartavionics, 2019-04-15:
I used god mode plugin and nowhere does that material name appear in the report.

smartavionics, 2019-04-15:
Anyway, it doesn't really matter. Let's assume that it was just some glitch.

closed 2019-04-15 by smartavionics