Ultimaker/Cura #5637: [4.0.0] line width / flow rate compensation factor / calibration cube fails single wall thickness check

quarky42, 2019-04-18:
**Application Version**
Released 4.0.0

**Platform**
Windows 10 x64

**Printer**
Snapmaker, 0.4mm nozzle, repeated the test / verified issue still present with a brand new nozzle that I installed today. 

**Project Files and Profiles:**
[Cura_CalibrationCubes_curaprofile.zip](https://github.com/Ultimaker/Cura/files/3095318/Cura_CalibrationCubes_curaprofile.zip)
[Cura_CalibrationCubes_Projects_3mf.zip](https://github.com/Ultimaker/Cura/files/3095319/Cura_CalibrationCubes_Projects_3mf.zip)

**GCODE:**
[Cura_CalibrationCubes_gcodes.zip](https://github.com/Ultimaker/Cura/files/3095323/Cura_CalibrationCubes_gcodes.zip)

**Steps to Reproduce**
Load a 25mm calibration cube into Cura 4.0.0  (and 3.6.0 for comparison on how the older version does line widths correcly)
Rotate models 45 degrees and resize only the z axis down to 10mm  (I only need about 10mm for measuring the single wall thickness of the calibration print.)
Set Cura to print 1 wall, 0.5mm thick (I use a thicker first layer and a higher flow rate for the first layer to help get good bed adhesion).   

**Actual Results**
**Cura 3.6.0** produces a gcode file that when printed gives me 0.51mm thick wall with the settings included on my old nozzle.  On my new nozzle, I get 0.71mm thick wall from Cura 3.6.0 and again 1.1mm thick walls from Cura 4.0.    (Measured with calipers.)

It appears that Cura 4.0.0 is ignoring the flow rate compensation factor.  Both prints appear to be single wall as far as I can see.  I believe when I reduce my flow rate compensation factor from 77% down to 53.4%, I will again have 0.5mm thick lines.  The nozzle was replaced because it got clogged at the end of my last print on the previous day.

This amounts to overextrusion of a little over 200% with Cura 4.0.0.  That doesn't make sense to me because the flow rate factor is only 77%.  So perhaps there is another source of overextrusion in Cura 4.0.0 in addition to this?

When Cura 4.0.0 first came out I was having this problem and switched back to using 3.6.0.  I tried Cura 4.0.0 again recently to try and reproduce this issue and submit it and I am finding the same thing is happening again with layers that are about twice as thick as they should be.

**Expected results**
I expect Cura 4.0.0 to produce walls that are roughly the same thickness as Cura 3.6.0 when using the flow rate compensation factor.   If Cura 4.0.0 cannot produce an accurate line width, then issues I have been having with it in terms of the accuracy of prints cannot be troubleshot properly.  I have printed larger more complex models that have movement features that have welded themselves together, and I have fitment issues with other models that I was not having with Cura 3.6.0 and I believe the issue is in not being able to print an accurate single wall line width.

quarky42, 2019-04-18:
Followup:   Since I put a new nozzle in, I went ahead and calibrated my e-steps again.  I was overextruding by 13%.  I adjusted my e-steps and reprinted the Cura 3.6.0 file that is attached above.   This time it came out 0.60mm thick walls using the existing gcode instead of 0.71mm from earlier.

I am reprinting the Cura 4.0.0 version, again, same version that is attached above and will measure the wall thickness there so you can see the difference.  It will likely be around 1mm thick instead of 1.1mm, but I will verify.

Ghostkeeper, 2019-04-23:
The profiles that you are using (or at least the ones you posted) are completely different between Cura 3.6.0 and 4.0.0. It looks like you modified them in 4.0 and didn't modify them in 3.6. The profiles are only ported to the new version when you start that version for the first time. They are not kept in sync.

For instance, in 3.6 you've set the initial layer bed temperature to 75 degrees, while in 4.0 you haven't overwritten the default. In 4.0 you've set it to retract before going to the outer wall, but not in 3.6. And most importantly, in 3.6 your g-code flavour is set to RepRap but in 4.0 your g-code flavour is set to RepRap (Volumetric). This last one should cause the feed to be interpreted as mm³ of material rather than mm of filament length, which would cause an overextrusion if your printer interprets it as mm.

closed 2019-04-23 by Ghostkeeper

quarky42, 2019-04-23:
I didn't set the printer to Volumetric in 4.0.0.   That is how it imported over.  Didn't even know that was an option.

I didn't set both Cura's up side by side at first.  I was using 3.6.0.  I tried to use 4.0.0 and got bad results.  I tried to figure out why.   I started by calibrating my extruder.  It was pretty far off.  No idea why I was getting good results in Cura 3.6.0 with such bad values.  I fixed that.   I printed a calibration cube with 3.6.0 and found that with a calibrated extruder, I had to set my extrusion calibration factor back to 77% to get a 0.5mm wall to come out as 0.52mm.    At that point I tried, what I believed, to be the same settings from 3.6.0 in 4.0.0 and found that I was getting over 1mm thick walls.   I went through the screens side by side and I set all the settings that I could see to be the same between them.  Sure there were still some residual differences that I didn't catch.   Can Cura 4.0.0 open Cura 3.6.0 projects?   If it can, I'll try that next time to help ensure that the project settings are the same.

I can see how digging through the files that are exported you exposed some settings that are different, but, what does the initial layer bed temperature to 75 degrees?   In both models the bed has to reach 75 degrees before it starts printing.  Also that wouldn't affect the wall width of a 10cm tall single wall.  (While I was printing a bunch of these calibration cubes, I was keeping the bed heated to 75 degrees manually anyway so it wouldn't cool back down and I wouldn't have to wait for it. However there was no way for you to know that by looking at the files.)

Neither of my printers are set to RepRap.  It was set to Marlin in 3.6.0 and it still is.   In 4.0.0 it somehow decided to choose Marlin (Volumetric) which must be some sort of import error as I didn't setup the printer in 4.0.0.  I just assumed that it had imported it over properly.  I'm glad you caught that.  I'll be glad to change that back to Marlin the way it is supposed to be and retry printing my calibration cube.  Maybe that's all the problem was.  However, if that is the case then the bug is that my printer that imported over to 4.0.0 during one of the betas (because that is when I first started trying 4.0.0) somehow decided to change the value.

Since you say it was set to RepRap in both, that also is potentially another problem.   I've started by fixing the setting in 4.0.0 to Marlin instead of Marlin (Volumetric).  I'll post the results positive / negative here as a followup.

quarky42, 2019-04-23:
As mentioned, I made the change from Marlin (Volumetric) to Marlin in Cura 4.0.0.  From 3.6.0 I saved a sliced gcode file, saved the project, saved the material, saved the printing profile and imported them all into Cura 4.0.0 where I sliced the object and saved the gcode file.  In 3.6.0 before I exported it, I reduced the flow rate to 75%.

This time I was able to get a 0.615mm (average) wall thickness on both the 3.6.0 and 4.0.0 models.  Thank you for helping me find that setting that was wrong.   I am at a loss for why my calibrated extruder is overextruding by about 23% at this point.   When I manually command it to extrude 100mm of filament, it gives me ~99.8mm or 99.9mm of filament extruded.

My filament is coming out at 1.75 and 1.76mm the places that I've measured it.

I can reduce my flow rate down to 60.5%.   If you have any thoughts on what I should try / where to look, I would welcome the suggestions.   I'll re-run the extruder calibration process later on today.

Ghostkeeper, 2019-04-24:
> Can Cura 4.0.0 open Cura 3.6.0 projects?

Yeah. It upgrades the 3.6 profiles in the same way as if it were installed.

The changes made to profiles between 3.6 and 4.0 were:

* Deleted the `bridge_wall_max_overhang` setting.
* Some stuff with saved network connections (Octoprint key, UM3/UMS5 network key).
* Increment `setting_version` to 6.

Other things, like g-code flavour, should not be changed.

> what does the initial layer bed temperature to 75 degrees?

Maybe 75 happened to be the default and you had manually overwritten it in one but not the other.

> Also that wouldn't affect the wall width of a 10cm tall single wall.

No, well, only the bottom few layers probably.

> Neither of my printers are set to RepRap. It was set to Marlin in 3.6.0 and it still is. In 4.0.0 it somehow decided to choose Marlin (Volumetric) which must be some sort of import error as I didn't setup the printer in 4.0.0.

Ah! Sorry, this one was my mistake. These g-code flavours were renamed at some point but their internal names didn't change. The internal name of `RepRap (RepRap)` gets displayed as "RepRap". The internal name of `RepRap (Volumetric)` gets displayed as "Marlin (Volumetric)".

> This time I was able to get a 0.615mm (average) wall thickness on both the 3.6.0 and 4.0.0 models. Thank you for helping me find that setting that was wrong.

Nice! Good news :smiley: Pretty hard to find, this one.

To debug whether Cura is wrong in the extrusion factor, you can pick any extruded line and calculate the length of it. Cura should extrude `length * line width * layer height` in material (mm³). So then depending on your firmware you just need to divide that by the surface area of your filament (`pi * 1.75mm^2`) to arrive at the correct E value, assuming that flow is 100%.

quarky42, 2019-04-24:
Sure, the first layer is a little wider.  I was ignoring that and measuring the other layers.  You can feel the "lip" of the first layer.  I was just using that as cheap insurance on good bed adhesion.  In both versions of cura the bed was heating up to 75 degrees before it started printing. It wasn't printing on a cold bed, ever.  I know because before every print, I catch the 20mm of extruded filament that is purged before the print starts. I don't like having that purge connected to my print.

Re: the flavors of gcode, thanks for explaining that.  That makes a lot more sense.  

Anyway, my latest test was to setup the same cube in Slic3r, this time trying 0.4mm thick walls.  The walls printed out at between 0.37mm and 0.41mm.    So about 0.39mm on average.  That is without any flow rate changes or anything. So, just comparing the two slicers, it seems like this has less to do with my printer and more to do with Cura.   I'll do some more testing with Slic3r.  I like the amount of controls that Cura gives me access to and I'm not a fan of Slic3r's interface in general so far.

Can you elaborate on what you were saying about picking any extruded line ?  I've reread that a few times and I'm not sure what you are saying there.  The math itself makes sense to me, but what do you mean "pick any extruded line"...  so you want me to make a 1 or 2 layer tall print and take the measurements?    I'm not sure I follow this line of logic though when I add my recent Slic3r test in.  It was able to print within +/- 5% of the commanded line width.

My Cura 0.40mm line width print test was 2 walls, so it should have come out as 0.80mm or even a little less if there was overlap / welding.   It measures out to be 1.15mm thick, instead.  Besides the line width problem, everything else about the print is great.  Nice sharp lines / edges, no creaking / cracking sounds (so good wall bonding)

Ghostkeeper, 2019-04-25:
> Can you elaborate on what you were saying about picking any extruded line ?

This is a way to see if the slicer is giving the correct amount of extrusion. I'll give an example. Take these two lines of g-code from halfway through a file that I recently sliced with an Ultimaker 3 on PLA's default profile:

    G1 F4178.6 X120.339 Y106.47 E81.71452
    G1 X112.66 Y108.528 E81.76686

If I want to calculate if the E value of the second line is correct, I first need to make it relative:

    81.76686mm - 81.71452mm = 0.05234mm

So it extruded ~0.05mm of filament to print this line. Since my filament has a diameter of 2.85mm, that has a surface area of `pi * r^2`, so:

    pi * (2.85mm / 2)^2 ~= 6.3794mm^2

The volume of material that Cura says we should be extruding is then that area times the length of filament, so:

    6.3794mm^2 * 0.05234mm = 0.333898mm^3

Now compare that with the volume of material that SHOULD have been extruded according to our desired line width and layer height.

The line width this was printed with is 0.42mm and the layer height was 0.1mm. The length of that line is obtained using Pythagoras via `sqrt(dx^2 + dy^2)`, so:

    sqrt((112.66 - 120.339)^2 + (108.528 - 106.47)^2) = 7.94999mm

This is a line of almost 8mm length. The expected volume of that line is width * height * length, so:

    7.94999mm * 0.1mm * 0.42mm = 0.333900mm^3

So it appears Cura is about 0.000002mm^3 off here, which is well within the realm of rounding errors. To me that looks fine.

If I were to have used a different flow rate than 100%, you'd have to take that into account as well. Also, if you've enabled wall overlap compensation in the settings you need to be sure to not take a wall line that overlaps with an adjacent wall.

quarky42, 2019-04-25:
Thank you!  The thing that was most unclear to me at first was where I was getting the values from.   I see what you are verifying within the gcode itself.   I will take that approach and try to compare gcode that Cura gave me which yields a single wall that is 24% too thick versus the gcode that Slic3r gave me which yields a single wall that is within a range of -7.5% under to +2.5% over the specified thickness. Hopefully this will give me more insight or at least help me narrow down where this difference is coming from.

Because I have an extrusion calibration factor of 0.75% in Cura and no such calibration factor in Slic3r, this really is confusing to me about where this difference in actual printed line width / single wall width is coming from as overlap wouldn't be an issue.

Thank you very much for helping me chase this down.