Ultimaker/Cura #5639: support eraser, per object setting tool has problems

joonghochoe, 2019-04-19:
**Application Version**

master version

**Platform**

windows 10 64bit

**Printer**

any printer

**Steps to Reproduce**

1. support eraser tool
    - open any model
    - select support eraser tool
    - change size or position eraser mesh
    - cura master version crash
2. per object setting tool
    - open any model for printing mesh
    - open any model for per object setting test
    - select per object setting tool to test mesh
    - select setting option
    - slice
    - change setting option of per object setting tool
    - don't change slice button status to slice ready

**Actual Results**

1. support eraser tool
    - cura master version is crashed
2. per object setting tool
    - slice button status is not changed

**Expected results**

1. support eraser tool
    - when change size, position of eraser mesh, cura is OK
2. per object setting tool
    - when change setting option of per object mesh, slice button is changed

**Additional Information**

If I revert def _onSettingChanged() in SettingOverrideDecorator.py, 1, 2 issues is cleared.
1. support eraser tool - if setCalculateBoundingBox is deleted
2. per object setting tool - if new*, changed variables is deleted

Ghostkeeper, 2019-04-23:
1. Reproduces here too. Pretty serious!
2. Indeed it should already start reslicing at the 4th step (if anything was sliced).

nallath, 2019-04-24:
Both of them are fixed for 4.1

closed 2019-04-24 by nallath

myloverby, 2020-08-15:
It says this is Solved? With Cura 4.1.. but I have 

--Cura 4.6 and the "eraser Tool" still Doesn't Work!--

I need to use it as a support blocker, tried every setting on the left, tried setting to "0 infill on the left menu" but it blocks nothing!!!

myloverby, 2020-08-15:
[image]
[image]
[image]

Ghostkeeper, 2020-08-17:
I think your confusion is that you expect the support "eraser" to actually erase support if you place it somewhere down lower in the support structure. While the support blocker tool actually only prevents support from being generated at all. It needs to intersect with the red parts of your model. If you place it down somewhere lower it won't prevent any support from being generated.

myloverby, 2020-08-17:
I figured it out, I just needed to set it to "normal mode" and set the
walls/infill all to zero on the support.