Ultimaker/Cura #5640: [4.0.0] Travel speed should not be limited

mdealer, 2019-04-19:
**Application Version**
4.0.0

**Platform**
Windows 10 x64

**Printer**
Custom FDM printer. CoreXY printer with an ultra-light print head.

**Steps to Reproduce**
Go to print settings and enter e.g. 800 mm/s to travel speed

**Actual Results**
Unable to slice if I enter 800. For whatever reason I can specify higher print speeds than travel speeds, which makes no sense.

**Expected results**
If there are any limitations, it should be obvious how to specify them. These should be per-printer and not hard-coded.

**Additional Information**
I'm running the printer on half power and it already can do 800mm/s, but the limitation within cura is way lower. There should be no arbitrary limitations. All min/max values should be configurable as part of the printer.

smartavionics, 2019-04-19:
You need to change the machine_max_feedrate_x and machine_max_feedrate_y values in your machine definition.

You can also alter these values directly in the GUI if you install the printer settings plugin from the marketplace.

Hope this helps.

Ghostkeeper, 2019-04-23:
I set the maxima to light speed now. They were used to determine the limits in the settings, which I think should not be limited unless someone knows the limits (e.g. if they are overwritten in the printer definition file). They were also used in the time estimates to get a better estimate if one of the axes was limited by the maximum feedrate but another is not. For instance you could be making a 3D move where the total speed gets limited by the Z axis which is typically much slower than the X and Y axes. However the default limits in fdmprinter weren't very realistic anyway.

commit 86336591 (2019-04-23): Set maximum feedrate to light speed

For many printers this maximum feedrate is unknown. The rationale for this commit is that if it's unknown we shouldn't limit the speed settings.
This does cause the time estimations to be out of whack if the speed is limited on one axis but not on another axis, such as if you home the build plate while homing the head as well (when the head is much faster than the build plate). Because now Cura doesn't assume that there is a limit on any axis any more. It just holds the print speeds.
Many printers override these settings so that they have a more realistic v [... cut]
commit ce9f3c9b (2019-04-23): Set maximum feedrate to light speed

For many printers this maximum feedrate is unknown. The rationale for this commit is that if it's unknown we shouldn't limit the speed settings.
This does cause the time estimations to be out of whack if the speed is limited on one axis but not on another axis, such as if you home the build plate while homing the head as well (when the head is much faster than the build plate). Because now Cura doesn't assume that there is a limit on any axis any more. It just holds the print speeds.
Many printers override these settings so that they have a more realistic v [... cut]
closed 2019-04-23 by Ghostkeeper