Ultimaker/Cura #5641: [4.0.0] Shadow not updated after reload of a model

thopiekar, 2019-04-19:
**Application Version**
4.0.0

**Platform**
Linux, but sure it doesn't matter

**Printer**
Doesn't matter

**Steps to Reproduce**
 * load an STL model [1]
 * make changes to that file
 * reload the file in Cura

**Actual Results**
Shadow doesn't match the new model [2]

**Expected results**
Shadow should match to the model

**Additional Information**
[1] First iteration
[image]
[2] After reload of second iteration
[image]

Ghostkeeper, 2019-04-24:
Unable to reproduce this. We recently did some work on the signals related to this though, so perhaps it has been fixed in master.

diegopradogesto, 2019-06-20:
I could reproduce it in 4.0, but not in 4.1. It's been magically fixed, then. @thopiekar could you please confirm?

closed 2019-06-20 by diegopradogesto