Ultimaker/Cura #5642: It uses a very long time when slicing tall models with some support blockers. Hardly any cpu usage

Silithas, 2019-04-19:
https://i.imgur.com/PpNwFJS.png

Cura hardly uses any cpu when slicing. and it doesn't even show up in programs running when i'm sorting by cpu usage.

Using latest 4.0

smartavionics, 2019-04-19:
Hi @Silithas , please save the project file, zip it and attach to this issue so we can investigate your issue. Thanks.

Ghostkeeper, 2019-04-24:
Not able to reproduce. Here is the model that I sliced: [tall_support_blocker.zip](https://github.com/Ultimaker/Cura/files/3110820/tall_support_blocker.zip)

Without support blocker this took 17.76860499382019s. Single sample.
With a support blocker this took 18.53211522102356s. Single sample.
It would stay at about 350% CPU usage throughout the slicing process on my computer, except that it dipped to about 25% halfway through support generation for a second, for both slices.

Silithas, 2019-04-25:
> Hi @Silithas , please save the project file, zip it and attach to this issue so we can investigate your issue. Thanks.

https://www.dropbox.com/s/idrmtshhnxgwswj/CE3_Gears1_V8E_PrintInPlace_BasicSupports.rar?dl=0

Ghostkeeper, 2019-04-29:
That RAR archive seems to be corrupt. My Archive Manager application can't open it, anyway. Is it a RAR or did you rename it?

Silithas, 2019-04-30:
Extracted fine here. Used winrar. Will zip it then instead of rar.

Silithas, 2019-04-30:
Also yes, it was a rar.

diegopradogesto, 2019-06-20:
The project file that you attached is only one object, and no support blockers are added. Do you mean "Tree supports" (because I saw you have that option enabled)? Actually when trying to slice it, it takes very long time, but when I disable tree supports, it's much faster. Checking the CPU usage, it's using almost 100%.

[image]

closed 2020-06-19 by no-response[bot]