Ultimaker/Cura #5644: Slicing Fails on many of my Models

bryancostanich, 2019-04-20:
Hi folks! 

I'm having some real troubles with the slicer in cura. Slicing doesn't complete on a lot of my .stl models I've created in AutoCad. It seems to finish, it says it's finished in cura; but when I go to print, the print just stops. I got in touch with LulzBot and they said it was due to .stl errors, and so I started to dig into it, and I no longer believe that's the case. It seems like it's a problem in the slicer. The .STL files open in and view just fine in a number of applications, including xcode, github's stl viewer, and some 3D apps I've tried. I've even tried to open the STLs up and "fix" them in sketchup, but often times the 3D fix plugins say there's nothing wrong.

So I think there's two issues here:

 1. Cura doesn't notify you if the slicer fails.
 2. The slicer is failing on these STLs

I've attached a problematic STL model, the printer config, and the project file.

**Application Version**
v3.2.32

**Platform**
macOS v10.14.4

**Printer**
LulzBot Taz 5

[LulzBot+TAZ+5+DualExtruder+v3.global.cfg.zip](https://github.com/Ultimaker/Cura/files/3100424/LulzBot%2BTAZ%2B5%2BDualExtruder%2Bv3.global.cfg.zip)
[Pogo_Mount_Bottom_PLA_Detail.3mf.zip](https://github.com/Ultimaker/Cura/files/3100425/Pogo_Mount_Bottom_PLA_Detail.3mf.zip)

[Meadow_F7_Micro_Pogo_Mount_Bottom.stl.zip](https://github.com/Ultimaker/Cura/files/3100429/Meadow_F7_Micro_Pogo_Mount_Bottom.stl.zip)

**Steps to Reproduce**
Try to slice the attached .stl. 

**Actual Results**
Slicing doesn't complete. If you try to print it will abruptly stop

**Expected results**
Slicing should finish, and if it doesn't finish, it should notify.

**Additional Information**

bryancostanich, 2019-04-20:
Also, FWIW, it looks fine in x-ray and layer view (no red):

[image]

[image]

smartavionics, 2019-04-20:
Hello @bryancostanich , I tried to open your project file in my Cura which is based on Ultimaker's source + my own tweaks and additions and it bombed so I can't test your profile but it slices your model just fine. I am not going to bother to print it as I know it will be OK.
Here's the layer view..

[image]

BTW, the stl file does have some problems as identified by FreeCAD...

[image]

Hope this helps.

bryancostanich, 2019-04-20:
Why do you suppose my print always stops then? I contacted LulzBot and they told me it's because the slicer has failed.

smartavionics, 2019-04-20:
I also checked the model with Cura's own tools and it reports...

[image]

> Why do you suppose my print always stops then? I contacted LulzBot and they told me it's because the slicer has failed.

Well, perhaps your Cura (which is derived from Ultimaker Cura but, in reality, is a completely different application) generates bad gcode when given a broken model. Basically, if the model is bad, all bets are off.

bryancostanich, 2019-04-20:
I can't seem to use Mesh tools with the Cura Lulzbot edition.  

In any case, I looked at the GCODE (attached), and it's complete through to the end. 

[Pogo_Mount_Bottom_PLA_Detail.gcode.zip](https://github.com/Ultimaker/Cura/files/3100516/Pogo_Mount_Bottom_PLA_Detail.gcode.zip)

fieldOfView, 2019-04-21:
MeshTools requires Cura 3.5 or newer.

The fact that an STL can be loaded and displayed does not mean that its geometry is suitable for 3d printing. Sketchup is not a tool that is suitable to repair a model for 3d printing.

bryancostanich, 2019-04-21:
> MeshTools requires Cura 3.5 or newer.

I upgraded to 3.6.2, and still don't see them. Maybe because it's the "Lulzbot" edition or whatever. 

> The fact that an STL can be loaded and displayed does not mean that its geometry is suitable for 3d printing. 

Sure, but if it's actually slicing, it should print using those gcode instructions, no? I viewed it in slice mode, and it's all there.

I looked at the GCODE output and it finishes, I see the end GCODE. But something in there is causing Marlin (the firmware) to actually reboot. :) What could cause that? I just took a glance through the GCODE and it looked normal, but I could have missed it. 

> Sketchup is not a tool that is suitable to repair a model for 3d printing.

There are a number of mesh repair plugins for Sketchup.

Ghostkeeper, 2019-04-24:
I'm not able to reproduce this issue. When slicing the 3MF file I do get these errors: 
```
2019-04-24 09:55:20,768 - DEBUG - [Thread-50] UM.Backend.Backend._backendLog [101]: [Backend] [ERROR] Clipper::offset failed. See generated debug.html!
```
That could indicate something wrong in the mesh, such as not being watertight, or being self-intersecting. I'm also not able to load your profiles though as the zip file seems to be corrupt.

Unless you're finding the M999 command (which is not generated by Cura by default), your printer should not restart. If your printer is restarting, that would be a firmware bug. The g-code generated on this side looks fine. You say that the g-code looks fine as well. The g-code is the only thing that is communicated between Cura and the printer, and the printer is doing something wrong as a result.

GregValiant, 2023-09-15:
Can we close this?

bryancostanich, 2023-09-18:
Sure.

closed 2023-09-18 by bryancostanich