Ultimaker/Cura #5645: [3.3.1] Missing most text after upgrade to Ubuntu 19.04

mtib, 2019-04-21:
**Application Version** 3.3.1

**Platform** Ubuntu 19.04

**Steps to Reproduce**
- Updated from Ubuntu 18.10 to 19.04
- Started cura

**Actual Results**
- Most text is missing
- Unusable

**Expected results**
- Text to show up

**Additional Information**
[image]
[image]
[image]

Some text as well as all keyboard shortcuts show up, but most UI text doesn't. Everything was fine before the upgrade to Ubuntu 19.04 and I haven't had this problem with any other application.

Startup log: [cura.log](https://github.com/Ultimaker/Cura/files/3101219/cura.log)

mtib, 2019-04-21:
Deleting the files in `.config/cura/3.3` leaves me with this on startup: 
[image]

mtib, 2019-04-21:
Couldn't fix that, but running the AppImage of 4.0 from the website works. So...

nallath, 2019-04-23:
The 3.3.1 version is probably from the PPA, which is not supported by us. Since the 4.0 app image is working, this issue will be closed.

closed 2019-04-23 by nallath

Ghostkeeper, 2019-04-24:
Works for me both from source and via the AppImage in 19.04.

mtib, 2019-04-24:
Then something must have gone wrong when upgrading from 18.10 to 19.04. Then again, I found a fix.