Ultimaker/Cura #5647: [4.0.0] More details about cost and weight of materials

moham96, 2019-04-21:
**Application Version**
4.0.0

**Platform**
Gnu/Linux

**Printer**
Creality Ender-3

**Steps to Reproduce**
slice any model while choosing a material that has it's cost setup

**Actual Results**
The output overlay shows the cost of the whole model 

**Expected results**
I think it would be useful to divide the cost, weight, and length of material used to infill, support and shells and top and bottom layers, details below.

**Additional Information**
showing details about the cost and used material weight of the support and infill can be really useful when optimizing a model for cost and weight, some models print fine with and without support but sometimes support can give the model better surface quality or increase the chance of successful print, having an idea of how much I'm wasting on support and infill can help in deciding to decrease the amount of support or maybe remove it all together, this is especially useful in large prints which already cost a lot.

Ghostkeeper, 2019-04-24:
I'm a bit worried that this tooltip would become a bit of a spreadsheet and you'd lose oversight.

Ghostkeeper, 2019-04-25:
We've discussed this among the development team. It could be a useful addition but we don't think it needs to get any priority. We'll keep the ticket open but don't plan on implementing it any time soon. Later.

Thisismydigitalself, 2019-05-01:
With your permission I'd like to bring back an old suggestion of mine and shove it just beneath this issue:
https://github.com/Ultimaker/Cura/issues/3514