Ultimaker/Cura #5648: [4.0.0] show toolhead movement in preview mode

moham96, 2019-04-21:
**Application Version**
4.0.0

**Platform**
Gnu/Linux

**Printer**
not important

**Steps to Reproduce**
slice any model and go to preview mode, you can see individual layers added on top of each other using the slider on the right, but you can't see how the layer is being deposited.

**Actual Results**
You can see individual layers added on top of each other using the slider on the right, but you can't see how the layer is being deposited.

**Expected results**
It would be useful to see the tool head as it moves to form a layer, this is helpful for example to know where exactly a layer begins to form and ends.

guestisp, 2019-04-23:
Just move the single layer slider (on the bottom side of the screen) to see the toolhead moving around

guestisp, 2019-04-23:
Or hit "play" in the same place

moham96, 2019-04-23:
@guestisp  maybe I'm missing something, I don't see any slide on the bottom or a "play" button !!
[image]

fieldOfView, 2019-04-23:
Your layer view is in "compatibility mode", probably because your GPU does not support the advanced OpenGL 4.1 functionality that the normal layer view uses. Compatibility mode does not have the horizontal slider to "play" the layers.

Ghostkeeper, 2019-04-24:
We don't intend to backport this feature to compatibility mode. While possible, the compatibility mode is currently just showing a bunch of polygons which is kind of easy.
OpenGL 4.0 got released in 2010. You'd expect most devices to be up to date by now...

closed 2019-04-24 by Ghostkeeper