Ultimaker/Cura #5649: [4.0.0-from-source] GUI/slicing error - "material specification = undefined"

agordon, 2019-04-22:
[image]
Hello,

I'm using cura built from git, and execute `cura_app.py` from the git directory.
Everything works well, except that after slicing, the material is "undefined",
and so the material length and cost are zero (see attached screen shot).

This does not happen when I run cura from an AppImage - so it's surely just a local configuration issue - but what am I missing? Any hints would be appreciated.

**Application Version**
Using cura from git, both tag "4.0.0" and latest master revision.

**Platform**
Debian 9.8/amd64.

**Printer**
Ender 3, but it's doesn't seem to be relevant, using other printers also shows "material specification" as undefined.

**Steps to Reproduce**
1. Deleted `$HOME/.config/cura`, `$HOME/.cache/cura`, `$HOME/.local/share/cura`
2. Go to the cura git direcotory, execute:
```
export PYTHONPATH=/home/gordon/sources/Uranium/:$PYTHONPATH
./cura_app.py
````
3. Cura loads, shows "welcome" screen
4. Add new printer (Creality Ender 3)
5. Use "generic PLA", change currency to "C$", set cost to 30, set weight to 1000g.
6. Load any STL file
7. Slice with default settings
8. slicing result is attached in screenshot.

**Actual Results**
"Material specification" is shown as "undefined"

**Expected results**
Show the non-zero length of material used, and corresponding cost.

Thanks!

Ghostkeeper, 2019-04-24:
Maybe some of the profiles are missing? You do have [fdm_materials](https://github.com/Ultimaker/fdm_materials) in there, right?

agordon, 2019-04-24:
Thanks for the reply.
Yes, there is a "material" sub-directory (cloned from "fdm_materials" repository) under the "resources" directory.
Other ideas?
Since I'm running from source, perhaps there's a way to enable more debugging / logging ?

thanks!

Ghostkeeper, 2019-04-25:
You could add some log entries to `plugins.CuraEngineBackend.CuraEngineBackend._onPrintTimeMaterialEstimates`. That's the function in the front-end that receives the time and material estimates from CuraEngine after slicing is completed.

By default Cura logs at the debug level already. That's all the logs we have.

agordon, 2019-05-02:
Thanks for the pointer.
In `_onPrintTimeMaterialEstimates`, the `message.repeatedMessageCount("materialEstimates")` used in the `for` loop is empty in my case - so `material_amounts` list remains empty.

If I add a dummy value, such as `material_amounts.append(5000)` after the `for` loop (line 721), the UI displays the material information (obviously, incorrect information based on 5000 mm^3).

I couldn't find where the "materialEstimates" messages are populated - any further hints?

Thanks!

agordon, 2019-05-02:
Update:
After pulling the latest git versions of both  [Cura](https://github.com/Ultimaker/Cura) and [CuraEngine](https://github.com/Ultimaker/CuraEngine) the problem went away - material estimates are now showing just fine.
Perhaps it was a mismatch/incompatible versions of my Cura vs my CuraEngine.

Ghostkeeper, 2019-05-03:
> I couldn't find where the "materialEstimates" messages are populated - any further hints?

They are populated in libArcus when it receives a message from CuraEngine. LibArcus is very stable as far as I'm concerned, so I'd expect that the real problem was that CuraEngine didn't send it for some reason. The fact that updating CuraEngine fixed it confirms this suspicion.

closed 2019-05-03 by Ghostkeeper