Ultimaker/Cura #5650: All at once error, offset?

BixlerVDL, 2019-04-22:
I have a problem with the all at once feature in cura.

I believe that my printhead settings are correct, so that they allow multiple objects to be printed at once. However, even small objects, smaller than the gantry height, will not allow Cura to slice.

I have tried to upgrade to Cura 4 from 3.6 with no improvement. I also tried to slice 1 big object, it worked with "all at once" but not with "one at a time". That made me think that there is an additional offset in "one at a time". 

Please let me know what the issue is. I want to print one at a time to improve overall print quality and to reduce time and stringing.
[image]
[image]
[image]
[image]
[image]

Thank you in advance!

Ghostkeeper, 2019-04-24:
"All at once" is the default mode of Cura. It makes the printer print the layers from bottom to top for all objects at the same time, rather than the first object first from bottom to top, then starting with the next object at the bottom again.

One at a time does have an extra border around each model to make sure that the print head doesn't collide with the previously printed models. Because each model is processed completely separately, the platform adhesion is also processed separately and so the grey areas around the build plate for brim/skirt/raft will also move to the models themselves. They may not collide.

Cura should allow you to slice as long as the broad shadow does not intersect with the small shadow of another object, and the object is lower than the gantry height of the printer (in order to not collide with the printer's gantry).

Normally any objects that don't fit will be left out of the print, as an easy way to disable printing some objects by moving them out of the build volume. With one-at-a-time mode this is not possible any more because of the way in which the printing order is determined.

Everything that can be seen in your screenshot looks expected behaviour, except the big one (second to last screenshot). The wide collision area goes outside the build volume, but that shouldn't prevent it from being printed. I can reproduce that bug as well.

nickstokes, 2019-05-10:
I am getting the same thing with Cura where if any part of the broad shadow falls outside of the build plate, it will refuse to slice, even if it is a single item. My printer is an Creality Ender 3.

rburema, 2019-05-15:
Seems like it's in the printer definitions: Some printers use/overwrite `machine_head_polygon` (Creality printers) others `machine_head_with_fans_polygon` (UM printers), some both (Prusa).

The 'easy' fix would be to just remove all '...with_fans...' from every text, so the beheviour will be the same as for the UM machines (which handle this correctly).

We did notice some other issues though, which may or may not intersect this one. We've decided to make internal ticket CURA-6522

evtrados, 2019-12-11:
Closing this issue, since the one at a time overlap bug has been resolved.
The fix will be in the next Cura release.

closed 2019-12-11 by evtrados