Ultimaker/Cura #5651: cura keeps crashing

EduardoOliveira, 2019-04-22:
**Application Version**
3.6 and 4.0

**Platform**
Windows 10

**Printer**
Anycubic i3 mega

**Steps to Reproduce**
Normal utilization, random actions

**Actual Results**
Stops working, even wen restarted.
It seems to corrupt some config file, only works when completely uninstalled and reinstalled.

**Expected results**
to not stop working

**Additional Information**
Log of a crashing boot
```2019-04-22 22:34:25,042 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin ConsoleLogger
2019-04-22 22:34:25,057 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin CuraDrive
2019-04-22 22:34:25,082 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [81]: Found CuraEngine at: C:\Program Files\Ultimaker Cura 4.0\CuraEngine.exe
2019-04-22 22:34:25,085 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin CuraEngineBackend
2019-04-22 22:34:25,095 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin CuraProfileReader
2019-04-22 22:34:25,104 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin CuraProfileWriter
2019-04-22 22:34:25,114 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin OBJReader
2019-04-22 22:34:25,126 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin OBJWriter
2019-04-22 22:34:25,142 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin STLReader
2019-04-22 22:34:25,153 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin STLWriter
2019-04-22 22:34:25,162 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin FileLogger
2019-04-22 22:34:25,175 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin FirmwareUpdateChecker
2019-04-22 22:34:25,186 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin FirmwareUpdater
2019-04-22 22:34:25,196 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin GCodeGzReader
2019-04-22 22:34:25,206 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin GCodeGzWriter
2019-04-22 22:34:25,217 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin GCodeProfileReader
2019-04-22 22:34:25,237 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin GCodeReader
2019-04-22 22:34:25,247 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin GCodeWriter
2019-04-22 22:34:25,260 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin ImageReader
2019-04-22 22:34:25,274 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin LegacyProfileReader
2019-04-22 22:34:25,284 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin LocalContainerProvider
2019-04-22 22:34:25,297 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin LocalFileOutputDevice
2019-04-22 22:34:25,308 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin MachineSettingsAction
2019-04-22 22:34:25,320 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin ModelChecker
2019-04-22 22:34:25,332 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin MonitorStage
2019-04-22 22:34:25,346 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin PerObjectSettingsTool
2019-04-22 22:34:25,359 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin PostProcessingPlugin
2019-04-22 22:34:25,369 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin PrepareStage
2019-04-22 22:34:25,379 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin PreviewStage
2019-04-22 22:34:25,397 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin RemovableDriveOutputDevice
2019-04-22 22:34:25,418 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin SimulationView
2019-04-22 22:34:25,433 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin SliceInfoPlugin
2019-04-22 22:34:25,445 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin SolidView
2019-04-22 22:34:25,459 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin SupportEraser
2019-04-22 22:34:25,484 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.getAccessToken [106]: No auth data to retrieve the access_token from
2019-04-22 22:34:25,489 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin Toolbox
2019-04-22 22:34:25,506 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin CameraTool
2019-04-22 22:34:25,522 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin MirrorTool
2019-04-22 22:34:25,538 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin RotateTool
2019-04-22 22:34:25,556 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin ScaleTool
2019-04-22 22:34:25,569 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin SelectionTool
2019-04-22 22:34:25,588 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin TranslateTool
2019-04-22 22:34:25,610 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin UFPWriter
2019-04-22 22:34:25,626 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin UltimakerMachineActions
2019-04-22 22:34:25,725 - DEBUG - [MainThread] UM3NetworkPrinting.src.Cloud.CloudOutputDeviceManager._onLoginStateChanged [57]: Log in state changed to False
2019-04-22 22:34:25,725 - DEBUG - [MainThread] UM3NetworkPrinting.src.Cloud.CloudOutputDeviceManager._onGetRemoteClustersFinished [80]: Parsed remote clusters to []
2019-04-22 22:34:25,726 - DEBUG - [MainThread] UM3NetworkPrinting.src.Cloud.CloudOutputDeviceManager._onGetRemoteClustersFinished [81]: Removed: 0, added: 0, updates: 0
2019-04-22 22:34:25,730 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin UM3NetworkPrinting
2019-04-22 22:34:25,751 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [36]: Checking for new version of cura
[... 129 lines cut ...]
2019-04-22 22:34:30,563 - DEBUG - [MainThread] cura.Machines.QualityManager.initialize [136]: Lookup tables updated.
2019-04-22 22:34:30,672 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [61]: Updating QualityProfilesDropDownMenuModel.
2019-04-22 22:34:30,673 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadContainer [59]: Loading container anycubic_i3_mega_draft
2019-04-22 22:34:30,678 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [386]: Container [anycubic_i3_mega_draft] added.
2019-04-22 22:34:30,817 - DEBUG - [MainThread] LocalContainerProvider.LocalContainerProvider.loadContainer [59]: Loading container anycubic_i3_mega_high
2019-04-22 22:34:30,822 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [386]: Container [anycubic_i3_mega_high] added.
2019-04-22 22:34:30,962 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [15]: Updating CustomQualityProfilesDropDownMenuModel.
2019-04-22 22:34:30,963 - INFO - [MainThread] cura.Machines.QualityManager.getQualityChangesGroups [179]: Cannot find node for machine def [anycubic_i3_mega] in QualityChanges lookup table
2019-04-22 22:34:31,417 - DEBUG - [MainThread] UM.Qt.QtRenderer._initialize [183]: Support for Vertex Array Objects: True
2019-04-22 22:34:31,425 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [99]: Initialized OpenGL subsystems.
2019-04-22 22:34:31,427 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [100]: OpenGL Version:  4.1.0 NVIDIA 418.91
2019-04-22 22:34:31,429 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [101]: OpenGL Vendor:   NVIDIA Corporation
2019-04-22 22:34:31,432 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [102]: OpenGL Renderer: GeForce GTX 980 Ti/PCIe/SSE2
2019-04-22 22:34:31,435 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 4.0\resources\shaders\default.shader]...
2019-04-22 22:34:31,440 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 4.0\resources\shaders\selection.shader]...
2019-04-22 22:34:31,444 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 4.0\resources\shaders\default.shader]...
[... cut ...]
2019-04-22 22:34:31,468 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 4.0\resources\shaders\default.shader]...
2019-04-22 22:34:31,472 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 4.0\resources\shaders\grid.shader]...
2019-04-22 22:34:31,476 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Ultimaker Cura 4.0\resources\shaders\platform.shader]...
```

Ghostkeeper, 2019-04-24:
Does it display a crash dialogue when it crashes, or does Cura just disappear entirely?

From the log, nothing in particular can be seen. I see only normal behaviour. The log entry seems to stop at the moment when the display is being drawn.

juri8, 2019-04-26:
On my machines (arch linux), it crashes when I click the _Preview_ button

fieldOfView, 2019-04-26:
@juri8, please try the AppImage. Yours is a different crash than @EduardoOliveira's.

EduardoOliveira, 2019-04-30:
all cura windows become "not responding" with a white overlay, and it takes about 2 kills to close the windows.

fieldOfView, 2019-04-30:
@EduardoOliveira do you have multiple screens?

EduardoOliveira, 2019-04-30:
I do, 1 screen 2k and another 1080p

fieldOfView, 2019-04-30:
On some configurations, Cura does not like starting up on a secondary screen. Cura remembers its window position, and if you closed Cura on the secondary screen that will cause it to reopen on the secondary screen. On most computers that is not a problem, on yours it is.

You can get Cura working again by pressing (and releasing) the windows key, typing (exactly, without the quotes) "%APPDATA%\Cura\4.0" followed by enter, and then editing the file named "cura.cfg". From the [general] section, remove all lines that start with "window_". Then save the file and start Cura again.

EduardoOliveira, 2019-04-30:
yep, I was able to reproduce the error consistently and fix it using the workaround.
In my case the lines that it created were:
```
window_left = 3228
window_state = 2
window_top = 405
```
any idea why it doesn't like my computer?

Thank you for the workaround

Ghostkeeper, 2019-05-01:
More suspiciously, why did it previously render fine on that secondary screen?

Maybe there would be a division by 0 somewhere if we ever divide something by the pixel scale factor? Though normally we only multiply by it. And you would think that QML is resistant to that sort of thing.

EduardoOliveira, 2019-05-01:
@Ghostkeeper, I've just tested the following:

- Open in main screen maximized
- "Un-maximized" it
- dragged it to the other screen
- maximized it
- sliced de model
- "Un-maximized" it
- dragged it back to the other screen
- maximized it

and no problem....

EduardoOliveira, 2019-05-10:
@Ghostkeeper I've seen you post on #5699 , if you need me to run a custom build or w/e let me know

Ghostkeeper, 2019-05-10:
No, well, for now I have no real ideas on how to solve this issue :confused:

Ghostkeeper, 2019-08-23:
Seems to be caused by this Qt bug: https://bugreports.qt.io/browse/QTBUG-69481

ajmilner85, 2020-04-01:
I'm still having this crash with Cura 4.5. Forgot to bring it back to my main monitor before closing it and now it crashes on startup trying to initialize on my secondary monitor.

2020-04-01 17:37:44,289 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin ConsoleLogger
2020-04-01 17:37:44,299 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin CuraDrive
2020-04-01 17:37:44,314 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [78]: Found CuraEngine at: C:\Program Files\Ultimaker Cura 4.5\CuraEngine.exe
2020-04-01 17:37:44,316 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin CuraEngineBackend
2020-04-01 17:37:44,323 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin CuraProfileReader
2020-04-01 17:37:44,324 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [173]: Socket state changed to Listening
2020-04-01 17:37:44,327 - INFO - [MainThread] UM.Backend.Backend.startEngine [81]: Started engine process: C:\Program Files\Ultimaker Cura 4.5\CuraEngine.exe
2020-04-01 17:37:44,329 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [101]: [Backend] Calling engine with: ['C:\\Program Files\\Ultimaker Cura 4.5\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '']
2020-04-01 17:37:44,334 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin CuraProfileWriter
2020-04-01 17:37:44,340 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin OBJReader
2020-04-01 17:37:44,346 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin OBJWriter
2020-04-01 17:37:44,354 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] 
2020-04-01 17:37:44,356 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin STLReader
2020-04-01 17:37:44,357 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] Cura_SteamEngine version 4.5.0
2020-04-01 17:37:44,359 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [177]: Socket state changed to Connected
2020-04-01 17:37:44,360 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] Copyright (C) 2019 Ultimaker
2020-04-01 17:37:44,361 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [167]: Backend connected on port 49674
2020-04-01 17:37:44,362 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] 
2020-04-01 17:37:44,364 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] This program is free software: you can redistribute it and/or modify
2020-04-01 17:37:44,365 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] it under the terms of the GNU Affero General Public License as published by
2020-04-01 17:37:44,366 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] the Free Software Foundation, either version 3 of the License, or
2020-04-01 17:37:44,368 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] (at your option) any later version.
2020-04-01 17:37:44,369 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] 
2020-04-01 17:37:44,370 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] This program is distributed in the hope that it will be useful,
2020-04-01 17:37:44,372 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2020-04-01 17:37:44,373 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2020-04-01 17:37:44,375 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] GNU Affero General Public License for more details.
2020-04-01 17:37:44,376 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] 
2020-04-01 17:37:44,377 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] You should have received a copy of the GNU Affero General Public License
2020-04-01 17:37:44,379 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2020-04-01 17:37:44,383 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin STLWriter
2020-04-01 17:37:44,389 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin FileLogger
2020-04-01 17:37:44,398 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin FirmwareUpdateChecker
2020-04-01 17:37:44,403 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin FirmwareUpdater
2020-04-01 17:37:44,409 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin GCodeGzReader
2020-04-01 17:37:44,415 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin GCodeGzWriter
2020-04-01 17:37:44,420 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin GCodeProfileReader
2020-04-01 17:37:44,432 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin GCodeReader
2020-04-01 17:37:44,437 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin GCodeWriter
2020-04-01 17:37:44,445 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin ImageReader
2020-04-01 17:37:44,450 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin LegacyProfileReader
2020-04-01 17:37:44,456 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin LocalContainerProvider
2020-04-01 17:37:44,463 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin LocalFileOutputDevice
[... cut ...]
2020-04-01 17:37:52,260 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.5\resources\shaders\grid.shader]...
2020-04-01 17:37:52,263 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [56]: Loading shader file [C:\Program Files\Ultimaker Cura 4.5\resources\shaders\platform.shader]...
2020-04-01 17:37:54,677 - DEBUG - [Thread-3] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM4 if baud rate 250000 works. Retry nr: 0. Wait timeout: 3
2020-04-01 17:37:59,180 - DEBUG - [Thread-3] USBPrinting.AutoDetectBaudJob.run [50]: Checking COM4 if baud rate 500000 works. Retry nr: 0. Wait timeout: 3

Ghostkeeper, 2020-06-29:
This is not fixed yet, mr. Bot. It should get fixed if we upgrade to 5.12 or newer.

GregValiant, 2024-11-15:
Is this still an issue with current Cura versions (5.8.0 and up).  Can this be closed?

EduardoOliveira, 2024-11-15:
Can't reproduce.

GregValiant, 2024-11-15:
I'll close it as it no longer seems relevant.

closed 2024-11-15 by GregValiant