Ultimaker/Cura #5653: [Cura 4.0 Feature Request] UX feedback when slice button clicked.

mapsedge, 2019-04-23:
**Application Version**
4.0

**Platform**
Windows 10

**Printer**
Flsun 3D Cube

**Steps to Reproduce**
Click the "Slice" button

**Actual Results**
Nothing that I can tell until - after some pause - Cura shows the slicing progress bar.

**Expected results**
What should happen is that there is some immediate UX feedback that the button has been clicked, something as simple as changing the button caption to "Please wait...". The pause between clicking and the progress bar showing up can be lengthy, long enough that sometimes I start to doubt that the click worked at all.

guestisp, 2019-04-23:
Totally agree

Ghostkeeper, 2019-04-24:
Duplicate of https://github.com/Ultimaker/Cura/issues/5480.

closed 2019-04-24 by Ghostkeeper