Ultimaker/Cura #5654: Preview window shows nothing

kingjamez, 2019-04-23:
**Application Version**

4.0.0
**Platform**

Windows 10 with AMD Radeon RX570 
**Printer**

Custom Delta
**Steps to Reproduce**

Slice anything, open any Gcode. 
**Actual Results**

The preview window shows pure white. No model or gcode visualization is displayed. The layer slider is there along with the play/pause slider. The play/pause slider will show that it's working but nothing is displayed. 
**Expected results**

The gcode from the slicing process should be displayed.
**Additional Information**

All other visualization works. I can import and position the object, I see the bed and bounding area of the printer.

Ghostkeeper, 2019-04-24:
I'm unable to reproduce this.

[image]

Maybe try going into the preferences and enabling the Compatibility Mode for layer view?

luukakkerman, 2019-05-01:
Same problem after just having installed Cura 4. 
Preview window doesnt show layer view.

Win10, Quadro K2200.

Ghostkeeper, 2019-05-03:
Could you try going into the Preferences and enabling Layer View Compatibility Mode?

Ghostkeeper, 2019-05-03:
Or maybe one of youse has a log file.

kingjamez, 2019-05-03:
Issue went away for me after I did the following:
Resized from full screen to windowed. 

After I did that, the preview showed up, and it still shows up when I go back to full screen.

closed 2019-05-03 by kingjamez

Ghostkeeper, 2019-05-06:
That's weird though. Maybe it didn't refresh the screen after rendering layer view?

luukakkerman, 2019-05-06:
> Same problem after just having installed Cura 4.
> Preview window doesnt show layer view.
> 
> Win10, Quadro K2200.

Fixed after restarting the fresh-installed Cura 4.0 app (or maybe even a system restart, cannot remember fully). Can see layer view now.