Ultimaker/Cura #5656: Slicing Process won't start

rtuz2th, 2019-04-23:
When I hit slice the progress bar stays at 0 forever. I tried reinstalling Cura and downloaded the most recent version, this made no difference.

My System: 

Win 10 64 Bit (10.0.17134) (No Antivirus installed, I tried disabling Windows Defender but this made no difference)

My Log:

2019-04-23 11:31:14,854 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin ConsoleLogger
2019-04-23 11:31:14,866 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin CuraDrive
2019-04-23 11:31:14,884 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [81]: Found CuraEngine at: C:\Program Files\Ultimaker Cura 4.0\CuraEngine.exe
2019-04-23 11:31:14,887 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin CuraEngineBackend
2019-04-23 11:31:14,897 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin CuraProfileReader
2019-04-23 11:31:14,907 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin CuraProfileWriter
2019-04-23 11:31:14,918 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin OBJReader
2019-04-23 11:31:14,927 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin OBJWriter
2019-04-23 11:31:14,941 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin STLReader
2019-04-23 11:31:14,952 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin STLWriter
2019-04-23 11:31:14,962 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin FileLogger
2019-04-23 11:31:14,976 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin FirmwareUpdateChecker
2019-04-23 11:31:14,988 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin FirmwareUpdater
2019-04-23 11:31:14,999 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin GCodeGzReader
2019-04-23 11:31:15,010 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin GCodeGzWriter
2019-04-23 11:31:15,022 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin GCodeProfileReader
2019-04-23 11:31:15,040 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin GCodeReader
2019-04-23 11:31:15,053 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin GCodeWriter
2019-04-23 11:31:15,066 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin ImageReader
2019-04-23 11:31:15,077 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin LegacyProfileReader
2019-04-23 11:31:15,090 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin LocalContainerProvider
2019-04-23 11:31:15,103 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin LocalFileOutputDevice
2019-04-23 11:31:15,116 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin MachineSettingsAction
2019-04-23 11:31:15,127 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin ModelChecker
2019-04-23 11:31:15,142 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin MonitorStage
2019-04-23 11:31:15,157 - WARNING - [MainThread] UM.Preferences.getValue [97]: Tried to get the value of non-existing setting cura/active_mode.
2019-04-23 11:31:15,160 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin PerObjectSettingsTool
2019-04-23 11:31:15,175 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin PostProcessingPlugin
2019-04-23 11:31:15,190 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin PrepareStage
2019-04-23 11:31:15,204 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin PreviewStage
2019-04-23 11:31:15,221 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin RemovableDriveOutputDevice
2019-04-23 11:31:15,241 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin SimulationView
2019-04-23 11:31:15,257 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin SliceInfoPlugin
2019-04-23 11:31:15,272 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin SolidView
2019-04-23 11:31:15,287 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin SupportEraser
2019-04-23 11:31:15,307 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.getAccessToken [106]: No auth data to retrieve the access_token from
2019-04-23 11:31:15,310 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin Toolbox
2019-04-23 11:31:15,328 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin CameraTool
2019-04-23 11:31:15,345 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin MirrorTool
2019-04-23 11:31:15,363 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin RotateTool
2019-04-23 11:31:15,381 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin ScaleTool
2019-04-23 11:31:15,397 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin SelectionTool
2019-04-23 11:31:15,415 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin TranslateTool
2019-04-23 11:31:15,441 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin UFPWriter
2019-04-23 11:31:15,459 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin UltimakerMachineActions
2019-04-23 11:31:15,546 - DEBUG - [MainThread] UM3NetworkPrinting.src.Cloud.CloudOutputDeviceManager._onLoginStateChanged [57]: Log in state changed to False
2019-04-23 11:31:15,547 - DEBUG - [MainThread] UM3NetworkPrinting.src.Cloud.CloudOutputDeviceManager._onGetRemoteClustersFinished [80]: Parsed remote clusters to []
2019-04-23 11:31:15,547 - DEBUG - [MainThread] UM3NetworkPrinting.src.Cloud.CloudOutputDeviceManager._onGetRemoteClustersFinished [81]: Removed: 0, added: 0, updates: 0
2019-04-23 11:31:15,550 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin UM3NetworkPrinting
2019-04-23 11:31:15,576 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin UpdateChecker
2019-04-23 11:31:15,581 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [36]: Checking for new version of cura
2019-04-23 11:31:15,611 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin USBPrinting
2019-04-23 11:31:15,631 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin UserAgreement
2019-04-23 11:31:15,653 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin VersionUpgrade21to22
2019-04-23 11:31:15,673 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin VersionUpgrade22to24
2019-04-23 11:31:15,693 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin VersionUpgrade25to26
2019-04-23 11:31:15,712 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin VersionUpgrade26to27
2019-04-23 11:31:15,732 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin VersionUpgrade27to30
2019-04-23 11:31:15,751 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin VersionUpgrade30to31
2019-04-23 11:31:15,772 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin VersionUpgrade32to33
2019-04-23 11:31:15,792 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin VersionUpgrade33to34
2019-04-23 11:31:15,813 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin VersionUpgrade34to35
[... 238 lines cut ...]
2019-04-23 11:32:17,573 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [227]: Previous socket existed. Closing that first.
2019-04-23 11:32:17,645 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [187]: Socket state changed to Listening
2019-04-23 11:32:17,675 - INFO - [MainThread] UM.Backend.Backend.startEngine [81]: Started engine process: C:\Program Files\Ultimaker Cura 4.0\CuraEngine.exe
2019-04-23 11:32:17,681 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [101]: [Backend] Calling engine with: ['C:\\Program Files\\Ultimaker Cura 4.0\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '']
2019-04-23 11:32:17,707 - DEBUG - [Thread-16] UM.Backend.Backend._backendLog [101]: [Backend] 
2019-04-23 11:32:17,713 - DEBUG - [Thread-16] UM.Backend.Backend._backendLog [101]: [Backend] Cura_SteamEngine version 4.0.0
2019-04-23 11:32:17,714 - DEBUG - [Thread-16] UM.Backend.Backend._backendLog [101]: [Backend] Copyright (C) 2018 Ultimaker
[... cut ...]
2019-04-23 11:32:18,191 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [432]: Sending slice message took 0.6201026439666748 seconds
2019-04-23 11:32:38,712 - DEBUG - [Thread-16] UM.Backend.Backend._backendLog [101]: [Backend] [ERROR] Could not connect to the given address
2019-04-23 11:32:38,806 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [811]: Backend quit with return code 0. Resetting process and socket.

Thank you very much for your help!

rtuz2th, 2019-04-23:
I just installed 3.6.0 and could reproduce the error. What am I doing wrong?

2019-04-23 11:55:52,639 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ConsoleLogger
2019-04-23 11:55:52,667 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [80]: Found CuraEngine at: C:\Program Files\Ultimaker Cura 3.6\CuraEngine.exe
2019-04-23 11:55:52,671 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin CuraEngineBackend
2019-04-23 11:55:52,682 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin CuraProfileReader
2019-04-23 11:55:52,693 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin CuraProfileWriter
2019-04-23 11:55:52,703 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin OBJReader
2019-04-23 11:55:52,715 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin OBJWriter
2019-04-23 11:55:52,732 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin STLReader
2019-04-23 11:55:52,744 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin STLWriter
2019-04-23 11:55:52,753 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin FileLogger
2019-04-23 11:55:52,769 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin FirmwareUpdateChecker
2019-04-23 11:55:52,781 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin FirmwareUpdater
2019-04-23 11:55:52,793 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin GCodeGzReader
2019-04-23 11:55:52,804 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin GCodeGzWriter
2019-04-23 11:55:52,815 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin GCodeProfileReader
2019-04-23 11:55:52,837 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin GCodeReader
2019-04-23 11:55:52,850 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin GCodeWriter
2019-04-23 11:55:52,864 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ImageReader
2019-04-23 11:55:52,876 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin LegacyProfileReader
2019-04-23 11:55:52,888 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin LocalContainerProvider
2019-04-23 11:55:52,902 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin LocalFileOutputDevice
2019-04-23 11:55:52,914 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin MachineSettingsAction
2019-04-23 11:55:52,926 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ModelChecker
2019-04-23 11:55:52,939 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin MonitorStage
2019-04-23 11:55:52,952 - WARNING - [MainThread] UM.Preferences.getValue [97]: Tried to get the value of non-existing setting cura/active_mode.
2019-04-23 11:55:52,955 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin PerObjectSettingsTool
2019-04-23 11:55:52,970 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin PostProcessingPlugin
2019-04-23 11:55:52,982 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin PrepareStage
2019-04-23 11:55:53,000 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin RemovableDriveOutputDevice
2019-04-23 11:55:53,024 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SimulationView
2019-04-23 11:55:53,040 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SliceInfoPlugin
2019-04-23 11:55:53,054 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SolidView
2019-04-23 11:55:53,069 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SupportEraser
2019-04-23 11:55:53,096 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin Toolbox
2019-04-23 11:55:53,112 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin CameraTool
2019-04-23 11:55:53,127 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin MirrorTool
2019-04-23 11:55:53,145 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin RotateTool
2019-04-23 11:55:53,163 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ScaleTool
2019-04-23 11:55:53,176 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SelectionTool
2019-04-23 11:55:53,194 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin TranslateTool
2019-04-23 11:55:53,218 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UFPWriter
2019-04-23 11:55:53,235 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UltimakerMachineActions
2019-04-23 11:55:53,312 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UM3NetworkPrinting
2019-04-23 11:55:53,329 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [36]: Checking for new version of cura
2019-04-23 11:55:53,332 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UpdateChecker
2019-04-23 11:55:53,377 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin USBPrinting
2019-04-23 11:55:53,392 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UserAgreement
[... cut ...]
2019-04-23 11:56:33,461 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [430]: Sending slice message took 0.6128029823303223 seconds
2019-04-23 11:56:40,657 - DEBUG - [MainThread] cura.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2019-04-23 11:56:54,108 - DEBUG - [Thread-13] UM.Backend.Backend._backendLog [94]: [Backend] [ERROR] Could not connect to the given address
2019-04-23 11:56:54,201 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [809]: Backend quit with return code 0. Resetting process and socket.

rtuz2th, 2019-04-23:
Okay, wiping down everything with revo uninstaller, going after usb drivers by hand and reinstalling everything then made it work. Sorry for spamming!

closed 2019-04-23 by rtuz2th

Ghostkeeper, 2019-04-24:
It looks like CuraEngine is not able to connect with the front-end. Maybe some firewall on your computer was blocking it and the re-install removed the quarantined file and replaced it with a non-quarantined one.