Ultimaker/Cura #5658: Alternative touchpad control

guestisp, 2019-04-23:
The current touchpad controls are very hard to use. They force users to use both keyboard and touchpad at the same time, thus they force users to use 2 hands, one on the keyboard, one on the touchpad

in example, shift+drag or ctrl+drag.
On multi-touch touchpad, you have to use at least 3 fingers: one for shift, one for clicking, one for dragging on the touchpad

Please simplify this or at least add an alternative touchpad option in the settings. The best config is the one from Slic3r Prusa Edition: click + drag : rotate the view, right lclick (two fingers on single button multi touch touchpad) + drag to move the view

free4ever, 2019-05-01:
I agree, the current touchpad controls are hard to use. I've switched over to Cura from Simplify3D, and its the one feature that I wish was available in Cura for multi-touch touchpad. Click + drag = rotate the view, and two finger + drag = move. I haven't used Slic3r but it appears to work the same way.  It just makes sense to control everything with one hand on the touchpad. Moreover an alternate control would make it a lot easier for people coming over from other slicers.

nallath, 2019-05-01:
The problem with this is that we also have a click & drag feature for models, which would then cause issues with rotating the camera. 

Being able to set different key combinations / shortcuts has been a feature we want for quite some time now (as it also makes it easier to sync workflow from CAD to Cura), but other things have gotten priority over this.

guestisp, 2019-05-01:
> I haven't used Slic3r but it appears to work the same way.

Yes, exactly. I use Slic3r as my primare slicer.

> Being able to set different key combinations / shortcuts has been a feature we want for quite some
> time now (as it also makes it easier to sync workflow from CAD to Cura), but other things have gotten 
> priority over this.

I understand, but the current touchpad control make Cura almost unusable (or very hard to use) on a multitouch laptop.

free4ever, 2019-05-01:
> The problem with this is that we also have a click & drag feature for models, which would then cause issues with rotating the camera.

I use the rotate and move for the view/camera a lot more often than moving the model. Instead of click + drag for moving the model, it can be ctrl + click + drag. When I have multiple models on the plate, I usually end using up Arrange All Models feature anyway.

The alternate touchpad controls can be disabled by default, and should be an option in the Preferences, and can be enabled if you want to use the alternate controls. This way it won't affect current users of Cura who are used to the existing controls.

guestisp, 2019-05-19:
Is this something being worked on?
I would like to use cura as my primary slicer but as i'm working with a laptop, the current touchpad controls are almost unusable

nallath, 2019-05-20:
It's not something that is currently being worked on. That, of course, won't mean that it will never be worked on.
If anyone feels up for the challenge, pull requests are welcome.

guestisp, 2019-05-20:
Thank you for the clarification. I'm not a Python developer, so I can't help. :(

guestisp, 2019-07-31:
Cura 4.2 is released. Any chance to get this fix on next release ? It's the only thing preventing me to use Cura, with a laptop and a multitouch touchpad it's almost impossible to properly use (and use easily) the current mouse controls, as we don't have the right button (we have to use multiple fingers and at the same time use the keybard... it's a mess)

nallath, 2019-07-31:
Nothing changed with regards to priority of this issue, sorry. I don't expect this to be in the 4.3 release.

guestisp, 2019-07-31:
Ouch. What a sadness

ncoquelet, 2020-01-11:
+1 Using cura control is a pain with a multi touch touchpad.
I understand that a key bind feature can resolve this issue, but for now we  just want that cura will be usable ...
Try by ourself, use control with a multi touch touchpad and you will understand our pain.

Other slicers, but also other  tools like fusion 365 or sketchup have behaviour which work very well with touchpad, so cura can do it too

Regards

kavehs87, 2021-10-16:
Can't use Cura because of this issue on my laptop (surface laptop).

dzid26, 2023-03-09:
The feature exists. It is Space+drag. No idea why can't it be also Ctrl+drag.