Ultimaker/Cura #5659: [4.0.0] If "Build plate shape" is "Elliptic", obj file is "too small to load" and is not loaded

Janaka490, 2019-04-23:
**Application Version**
Cura 4.0.0

**Platform**
 Windows 10 pro

**Printer**
Delta bot 

**Steps to Reproduce**

[Progetto CURA.zip](https://github.com/Ultimaker/Cura/files/3109422/Progetto.CURA.zip)
1) Setting - printers - manage printers - Select the printer in use, "Machine settings" - "Printer" "Build plate shape". Select "Elliptic".
2) File - open file.obj

**Actual Results**
3) A warning appears: "Warning The selected model was too small to load". 
4) Conseguence: The file is not loaded.

**Expected results**
Obj file should be opened

**Additional Information**
Workaround:
1) "Build plate shape". Select "Rectangular".
2) Open the file . No error
3) "Build plate shape". Select "Elliptic".

Maybe this issue is connected to #2139 "Model is not placed in center position but it can fit to the build volume"

Ghostkeeper, 2019-04-24:
I think I need to have that file.obj that you loaded. The project file that you uploaded is empty but it does contain settings for your "Deltabot by gms" printer. So I loaded my own cube.obj and it loaded fine.

Janaka490, 2019-04-30:
[laura04.zip](https://github.com/Ultimaker/Cura/files/3133092/laura04.zip)
File attached. 
When I load it with build shape "Rectangular", the message "Scaling object - auto scaled to 1000000% of original size" appears.
Please note that this model in Ultimaker cura 15.04.2 was imported without need to be scaled.

Ghostkeeper, 2019-05-01:
Yes! It does reproduce with that one.

The problem seems to be that Cura creates a collision hull around the model (in 3D) which is then flattened to produce the shadow and also used to determine the overall size more quickly. This convex hull computation has a limited precision and in your case produces no volume.

Maybe there is an underlying problem that the auto scaling doesn't work for elliptical build volumes?