Ultimaker/Cura #5660: Cura don´t open

AmanteVero, 2019-04-24:
**Application Version**

Version: 2.3.1

**Platform**

Platform: Windows-7-6.1.7600-SP0
Qt: 5.5.1
PyQt: 5.5.1

**Printer**

Geeetech A10
**Steps to Reproduce**

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 61, in <module>
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Application.py", line 310, in getInstance
  File "D:\2.3\build\inst\lib\python3.4\site-packages\cura\CuraApplication.py", line 173, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\QtApplication.py", line 94, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 107, in upgrade
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 281, in _upgradeFile
  File "C:\Python34\lib\codecs.py", line 319, in decode
MemoryError
**Actual Results**

**Expected results**

**Additional Information**

fieldOfView, 2019-04-24:
Quick fix: Go to C:\Users\<your username>\AppData\Local\cura and remove the file named Cura.log

Better fix: use a newer version of Cura (but you will need a 64 bit version of Windows for that)

Ghostkeeper, 2019-04-24:
Fixed in version 2.4.0: https://github.com/Ultimaker/Uranium/commit/6bea040f6b9584d94bd2d4acd87645be1ae05cdf

closed 2019-04-24 by Ghostkeeper