Ultimaker/Cura #5662: [4.0.0] support blocker generates too large print area

StilTeg, 2019-04-24:
**Application Version**
4.0.0

**Platform**
Windows 10 64bits

**Printer**
Ultimaker 3 Extended

**Steps to Reproduce**
Add support blocker, slice, sent to printer

**Actual Results**
 error message appears in Cura connect : print too large for printer

**Expected results**

**Additional Information**
 
This can be seen when simulating first layer, at the end of the layer, the printhead moves to the upper right corner where it should not
> [Cura Project](http://stilteg.legoutdurisque.fr/Docs/UM3E_SleepingAngel.3mf.zip)

Ghostkeeper, 2019-04-25:
Looks to me like it's making this move in order to prime the second extruder, the first time that that extruder gets used. It's a bit of a blind move 10mm towards the back. Maybe we need to make sure that the current position is not within 10mm of the back of the printer before making that move?

Ghostkeeper, 2019-04-25:
It doesn't seem to have anything to do with the support blockers. Removing them has the same problem.

StilTeg, 2019-04-25:
It does. Trying before adding any support blocker works

closed 2024-12-08 by github-actions[bot]