Ultimaker/Cura #5664: Cura 4.0 now stops after heating.

Baza-1969, 2019-04-25:
-->

**Application Version**
Cura 4.0

**Platform**
Windows 10

**Printer**
Anycubic Kossel Delta Linear Plus

**Steps to Reproduce**
Replaced extruder fan - now Cura stops after heating - connected by USB

**Actual Results**
When I select Print via USB - the bed heats, the extruder heats - all normal. The printer then performs the home position, and the extruder motor cycles.
 'printing..' is displayed on the LCD screen.
The printer does not move down to print - it just remains and the home position.

**Expected results**
The printer should heat, home, then move down to print.

**Additional Information**
I have also performed a PID autotuned to ensure that the extruder temp is stable (I replaced the extruder fan - hence the reason for PID tune)
I have also saved the file as Gcode and placed on an SD card - this prints perfectly - so the issue is USB related only. I have printed by USB for weeks with no issues.
One strange thing I've noticed is a '0' line in the Gcode usually near the start - around 'M117 Printing..' (please see attached)

[AKP_Testing_Circle_40301.zip](https://github.com/Ultimaker/Cura/files/3115521/AKP_Testing_Circle_40301.zip)

smartavionics, 2019-04-25:
> One strange thing I've noticed is a '0' line in the Gcode usually near the start - around 'M117 Printing..' (please see attached)

Looks like the post-processing has removed the layer number comments and just left the layer numbers which maybe are not acceptable in gcode?

Ghostkeeper, 2019-04-25:
That `0` line might cause the printer to throw up, indeed. I'm unable to reproduce that though. I don't get the `0` files unless I explicitly change lines to `0` with the search-and-replace script.

Maybe a project file would help.

Baza-1969, 2019-04-25:
I've also no idea how the '0' arrived in the gcode - sorry, how do I get a project file - I'm quite new to Cura?

Baza-1969, 2019-04-25:
@Ghostkeeper - you gave me a huge clue!
I was actually using the search / replace function earlier this week. I had entered a search phrase, but had left the replace field blank! - evidently this is enough to replace the searched item with a '0' !
Since I have cleared all the scripts, everything works again... many thanks for your help!

closed 2019-04-25 by Baza-1969