Ultimaker/Cura #5666: 4.0 - Restore from cura backups isn't working

supernovae, 2019-04-25:
Restoring settings is crashing and not working correctly. When you start after clicking restore it warns that it needs to reset profile and forces me to login and setup printer again. 

Cura 4.0
Windows 10 1809
MonoPrice Select V2 & HEVO 

**Steps to Reproduce**
I rebuilt my PC - formatted & re-installed win10 fresh and Cura 4.0.0 fresh. Clicked cura, chose to login, succesfully logged in and I clicked extensions -> cura backups -> clicked on my latest backup and clicked "Restore" - app says it will exit but doesn't exit - just kind of gets stuck. I click close and then restart cura and then it says 

"Configuration Errors
Your configuration seems to be corrupt. SOmething seems to be wrong with the following profiles: 
- Monoprice select mini v2 (e3d) 2
- monoprice select mini v2 (e3d) 2_user
Would you liek to reset factory defautls? Reset will remove all your current printers and profiles (Reset) 

(no other option besides reset)

**Actual Results**
Can't recover from online backup

**Expected results**
Restore settings successfully

**Additional Information**
Fresh OS install with Cura 4.0 fresh install - logged in and clicked restore settings.

supernovae, 2019-04-25:
It may have been a service error or backup error too - i chose an older one to restore and finally got a good restore after many failed attemps

supernovae, 2019-04-27:
Now that i've recovered an older backup, i have to re-sign in every time i launch when i don't believe i had this problem before.

Ghostkeeper, 2019-04-29:
The problem that you need to sign in again after every launch has been fixed in 4.1. We saw that too.

I think the back-up that you made had configuration errors at the time you made the back-up though, if an older one worked fine. Unless that back-up was made in 3.6?

closed 2020-07-03 by no-response[bot]