Ultimaker/Cura #5668: CURA UI Slow. Not my computer

Wdfitzpat007, 2019-04-26:
What ever info you need I will provide. Sorry if repost, I am too busy to look for an existing post.

I have a 2013 macbook pro 15" retina version, 4th gen intel hyper-threaded quadcore with 16 GB of ram and a SSD with 51gb of free space. Running out of memory/CPU time is not an issue. Bootcamp keeps my drivers up-to-date.

Fusion360, solidworks and other programs with intensive UI / graphics work fine. Inside cura everything works except the UI, for example I click on settings and it just hands for a few seconds at 18% CPU usage with an outline of the dropdown that should pop up. I have not tried re-installing, though will if you like. I can post any screenshots or video if you need. I expect this will take some effort on my end and hope that the users of my programs put in the same effort I will put into helping you.

Also, movement is funky. IE when I hover over the rotate circles or move arrows sometimes it fails to trigger a color change and thus nothing happens with click+drag.

**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

fieldOfView, 2019-04-26:
Duplicate of https://github.com/Ultimaker/Cura/issues/4505
Should be fixed in the upcoming Cura 4.1. A patched version of Cura 4.0 can be found here:
https://github.com/Ultimaker/Cura/issues/4505#issuecomment-482270203

closed 2019-05-10 by nallath

ilium007, 2019-05-30:
Cura 4.1 lagging for me. Fresh install. Unuseable.

nallath, 2019-05-31:
Look, only posting a "Me too" comment isn't helpful at all. If you don't post information about what your setup is, what the logs say and what it is that you're doing **exactly**, there is nothing we can do to fix it.

ilium007, 2019-06-04:
> Look, only posting a "Me too" comment isn't helpful at all. If you don't post information about what your setup is, what the logs say and what it is that you're doing **exactly**, there is nothing we can do to fix it.

What else do you want me to add ? Its a slow UI problem.

OSX 10.14.1
16GB RAM
i5 Mac Mini

What I'm doing exactly? Anything in the menus / UI.

What logs *exactly* do you want?

Ghostkeeper, 2019-06-04:
> What logs *exactly* do you want?

The Cura log could be helpful. Here's instructions on where to find it: https://github.com/Ultimaker/Cura#logging-issues