Ultimaker/Cura #5669: [4.0.0] Layer path is not smooth

kolczynsky, 2019-04-26:
**Application Version**
4.0.0

**Platform**
Windows 10

**Printer**
Creality Ender 3

**Steps to Reproduce**
Nozzle: 0.4 mm
Layer height: 0.12 mm
Wall thickness: 1.2 mm

**Actual Results**
After slicing, outer wall is not smooth everywhere as it should be. At lower layers it looks like this:

[image]

But on upper layers it is already fine:

[image]

Below is the sliced model. As you can see, the issued part is simple cilindrical shape, so layers should look the same on whole item.

[image]

**Expected results**
Here is how it is properly sliced in third-party slicer:

[image]

**Additional Information**
This glitch is not caused by seam, because I moved the seam to some other point. I've tried to resolve the problem by playing with "Outer Wall Inset" option, but it does not help.

At the end you can compare the print results. Cura's one is on top. Red arrows mark the glitches. At the bottom is a print prepared in other slicer. Print settings (material, temperature, layer height, nozzle diameter etc.) were the same in both cases.

[image]

If you find .3mf file is useful, here it is:

[.curaproject.zip](https://github.com/Ultimaker/Cura/files/3120812/default.curaproject.zip)

smartavionics, 2019-04-26:
Hello @kolczynsky. Unfortunately, your .3mf file only contains the model and not the settings. So please do a File -> Save (not export) and the resulting .3mf will contain both the model and the settings. Thanks.

kolczynsky, 2019-04-26:
My mistake :) As requested:

[.curaproject.zip](https://github.com/Ultimaker/Cura/files/3121335/default.curaproject.zip)

smartavionics, 2019-04-26:
Those glitches are being produced by the wall overlap compensation which is long known to have problems. If you disable that, the layer view looks nicer. Whether it will print nicer or not is another matter. Hope this helps.

[image]

kolczynsky, 2019-04-26:
Thank you! Disabling 'Compensate Wall Overlaps' resolved the problem completely. Printed part looks great now:

[image]

 As I remember 'Compensate Wall Overlaps' is hidden and enabled by default. So if it's known that it causes issues, maybe it would be a good idea to disable it by default in future Cura releases?

smartavionics, 2019-04-26:
> So if it's known that it causes issues, maybe it would be a good idea to disable it by default in future Cura releases?

But for a lot of models it is beneficial so even though flawed, it's probably better to keep it enabled by default.

kolczynsky, 2019-04-27:
Sure. Thanks a lot!

closed 2019-04-27 by kolczynsky