Ultimaker/Cura #5670: Bottom Hole filled only on multi-model

FHSES, 2019-04-26:
**Application Version**
 4.0.0

**Platform**
Win10

**Printer**
BQ Witbox2
[BQW2_4x_Custom_EmptyD6.zip](https://github.com/Ultimaker/Cura/files/3121588/BQW2_4x_Custom_EmptyD6.zip)
[BQW2_1x_Custom_EmptyD6.zip](https://github.com/Ultimaker/Cura/files/3121589/BQW2_1x_Custom_EmptyD6.zip)

**Steps to Reproduce**

When i slice 1 model the bottom hole is supported
If I multiply the model the bottom holes are filled!

**Actual Results**

The parts to print is wrong

**Expected results**

The parts to print should good

**Additional Information**

Even with 1 part it is hard to find the parameters to not have the bottom hole filled

smartavionics, 2019-04-26:
Hello @FHSES , I think the problem is with the model as the Cura meshtools finds problems...

[image]

BTW, you can probably save a little time when printing by using an Infill Layer Height of 0.2 (or maybe even 0.3).

FHSES, 2019-04-29:
Hello,
If i put only one model, there is no problem with the bottom holes!
[image]

But if i only multiply the model... no more hole at the bottom! 
[image]

and not depending of witch face is down (Customizable Dice)
[image]

smartavionics, 2019-04-29:
All true, but that doesn't get away from the fact that the model itself has issues. Fix the model first and then the slicer's performance can be evaluated.

FHSES, 2019-04-29:
> BTW, you can probably save a little time when printing by using an Infill Layer Height of 0.2 (or maybe even 0.3).

Yes Sure! Depending of the quality expected!

FHSES, 2019-04-29:
> the fact that the model itself

You right! they were issues in the model (flipped faces)
But I don't understand why the problem was solved with 1 model and not with multiples.

Thank You

closed 2019-04-29 by FHSES

Ghostkeeper, 2019-04-29:
Indeed, you can also see that it's not watertight in X-ray view:
[image]

If the model is not watertight, then technically there is no inside to your model. A human can understand that the volume that is 99% surrounded by triangles is probably meant to be filled with plastic, but a computer doesn't. Cura tries to close such small gaps at a best-effort attempt, but it is not reliable because there is no real solution to this.

The reason why it suddenly stops working with multiple meshes is because then there are multiple outlines on one layer and the whole problem of closing every single outline separately becomes more complex.