Ultimaker/Cura #5671: [current master] Unable to slice, multiextruder, key error

paukstelis, 2019-04-27:
I've recently pulled/rebuilt all master branches of necessary repos. When attempting to slice in a multiextruder environment (5 extruders) I get the following error:

```
2019-04-27 09:07:53,247 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [225]: Starting to slice...
2019-04-27 09:07:53,248 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [243]: Going to slice build plate [0]!
2019-04-27 09:07:58,966 - ERROR - [Thread-3] UM.Logger.logException [84]: Exception: Job <CuraEngineBackend.StartSliceJob.StartSliceJob object at 0x7f5576b11160> caused an exception
2019-04-27 09:07:58,983 - ERROR - [Thread-3] UM.Logger.logException [88]: Traceback (most recent call last):
2019-04-27 09:07:58,987 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/home/paul/git/Uranium/UM/JobQueue.py", line 121, in run
2019-04-27 09:07:58,992 - ERROR - [Thread-3] UM.Logger.logException [88]:     job.run()
2019-04-27 09:07:59,001 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/home/paul/git/Cura/cura/../plugins/CuraEngineBackend/StartSliceJob.py", line 263, in run
2019-04-27 09:07:59,013 - ERROR - [Thread-3] UM.Logger.logException [88]:     self._buildExtruderMessage(extruder_stack)
2019-04-27 09:07:59,021 - ERROR - [Thread-3] UM.Logger.logException [88]:   File "/home/paul/git/Cura/cura/../plugins/CuraEngineBackend/StartSliceJob.py", line 378, in _buildExtruderMessage
2019-04-27 09:07:59,029 - ERROR - [Thread-3] UM.Logger.logException [88]:     settings = self._all_extruders_settings[str(extruder_nr)].copy()
2019-04-27 09:07:59,031 - ERROR - [Thread-3] UM.Logger.logException [88]: KeyError: '5'
2019-04-27 09:08:00,934 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [188]: Error check finished, result = False, time = 11.3s
2019-04-27 09:09:25,635 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [51]: No more on-app-exit callbacks to process. Tell the app to exit.
```
Not sure if this error key is from StartJobResults (incompatible material; all materials are set as PLA) or if this is referencing something else. 

 Version**

Master

**Platform**

Linux - Ubuntu

**Printer**

**Steps to Reproduce**

Pull/build all master branches

**Actual Results**

Error thrown

**Expected results**

Slicing

**Additional Information**

paukstelis, 2019-04-27:
This happens even if I use a single extruder. It happens if I use an imported printer profile, or if I define a new custom printer with 5 extruders.

Ghostkeeper, 2019-04-29:
I'm not able to reproduce this on my development environment (Ubuntu 19.04). It looks a bit like there is a mistake in how the extruder definitions reference the printers or vice-versa, but it's hard to tell (and I wouldn't know how that could happen only for you).

Maybe you could run a bisect, if updating your clone started the issue?

paukstelis, 2019-04-29:
I traced it back to https://github.com/Ultimaker/Cura/commit/0d8b625bf8acea9c32156d7779a31289360d5389

The commits before that point work for me.

Ghostkeeper, 2019-04-30:
All right, I'll discuss today with Nallath. I think we may revert that commit then.

Ghostkeeper, 2019-04-30:
I could reproduce the issue now with the model posted here: https://github.com/Ultimaker/Cura/issues/5676: [Cura4badfloor.zip](https://github.com/Ultimaker/Cura/files/3130112/Cura4badfloor.zip)

nallath, 2019-04-30:
Fixed it! Thanks for investigating.

commit 3f47ffea (2019-04-30): Use only the active extruders when sending slice message

Fixes #5671
closed 2019-04-30 by nallath