Ultimaker/Cura #5672: Firmware retraction should be selectable per print profile

mdealer, 2019-04-28:
**Application Version**
4.0.0

**Platform**
Windows 10 x64

**Printer**
Custom FDM printer

**Steps to Reproduce**
Edit any profile.

**Actual Results**
 There is no option for firmware retraction.

**Expected results**
An option for firmware retraction which is used for all retractions. 

**Additional Information**
Other options which conflict with this option can be disabled. I also noticed there were other tickets regarding this, but their status is bogus, thus I created a new issue. The "solution" to use a plugin is a workaround. As long as the plugin exists, this issue cannot get old.

nallath, 2019-04-29:
Do you want to be able to change the firmware_retraction setting in the panel where you modify your custom FDM printer or for every profile?

If it's the second option, we won't support that as it's a machine specific setting.

Ghostkeeper, 2019-04-29:
Why would you want to vary this from print to print, above just from printer to printer?

mdealer, 2019-04-29:
I think it wouldn't matter where it is. I would want to vary this because there are implications for other settings, such as Z hop, wipe, coast, retraction, unretraction and relevant speeds. These settings are already in the print profiles but they are attributable to the printer.

Not to throw the setting concept upside-down, but every single setting should be ideally configurable per printer and per profile, including min, max, default and changeability. This is how I understand a setting and a set of settings.

nallath, 2019-04-29:
Those settings are in the profiles because they impact quality (and thus, could / should be set per profile. Some materials do / don't need wipe for instance). Whether the machine needs to use firmware retraction is machine specific (and not quality specific). Would you also want to change the size of the build plate in a profile?

smartavionics, 2019-04-29:
Here's my take on this. I used to use non-firmware retraction for all my printing but as the controllers I use support G10/G11. I have now moved to using firmware retraction. One gotcha is that although the controllers support the configuration of retract/prime lengths and speeds, Cura will still command extra extrusion after a G11 if the extra prime setting is non-zero when firmware retraction is in use. So there's not a clear division of labour between slicer and firmware when firmware retraction is enabled.

Also, it could be argued (err, suggested?) that the retraction and prime speed settings are hidden when firmware retraction is enabled as they are no longer effective.

Personally, I think the firmware retraction setting should at the very least be visible without having to have the printer settings plugin installed. Whether it should be per-profile, I am agnostic.

mdealer, 2019-04-29:
Bed size does not affect print quality, but retraction does. The machine does not need firmware retraction, the machine merely offers it. I can change retraction settings without repeating a lengthy slicing and upload process. 

What a machine needs is not to be defined by Cura.

nallath, 2019-04-29:
What a machine needs is something that should be known to Cura though ;) 
 
But in that case, I do understand the issue now. The setting seems to imply more things than just one thing. There should be a machine setting for "Is this machine capable at all of handling firmware retractions" and a non-machine setting for "Should it?" 

If the machine retraction is used, a bunch of other settings should at that point become hidden.

mdealer, 2019-04-29:
Regarding hidden stuff: I'd prefer disabling the affected options with an informative tooltip. As you can see in the discussion, it's not immediately clear what is affected and why.

smartavionics, 2019-04-29:
And not forgetting z-hop. That too can be controlled by the printer firmware during processing of G10/G11. Even though the printer I use can do that, I am still setting z-hop in the slicer.

So to summarise, the printer may or may not support g10/g11, and if it does, it may or may not be able to control extra prime amount and/or z-hop. Cura should be able to cope with those variations.