Ultimaker/Cura #5675: 4.0.0 Sliced model is incomplete

bkrahmer, 2019-04-29:
**Application Version**
4.0.0

**Platform**
Ubuntu 18.04

**Steps to Reproduce**
Open the attached file.  Note that slicing it in the default orientation looks fine.  Now rotate it 90 degrees on the red axis.  When I slice it, regardless of support settings, only a few layers of parts of the model are being sliced corrected.  I double-checked the file with slic3r, and it looks fine there.  The number of layers seems almost correct (16mm high x 0.2 layer height = 80, and I'm getting 86 layers), but the screenshot shows the preview slider on the top layer.  It feels like a bug.  

[bowl.zip](https://github.com/Ultimaker/Cura/files/3128864/bowl.zip)

[image]

smartavionics, 2019-04-29:
The model has problems.

Cura says...

[image]

Freecad says...

[image]

bkrahmer, 2019-04-29:
Will you please tell me how you saw that message in cura?  I didn't get any warnings.  I know that the model is not watertight, as it's half of an object, but I don't understand what that has to do with it.  Conceptually, that model is printable (as I mentioned, slic3r seems as if it will print it).  Can you shed any light on the subject?  Thanks!

smartavionics, 2019-04-29:
Install the mesh tools plugin from the marketplace.

[image]

Sorry, I don't know much about models except that when they are invalid they don't print correctly. Good luck!

Ghostkeeper, 2019-04-30:
Yeah, the entire top surface is missing. Cura can't know what is inside the model or outside, then.
Fixed it for you: [bowl.zip](https://github.com/Ultimaker/Cura/files/3130159/bowl.zip)

closed 2019-04-30 by Ghostkeeper

bkrahmer, 2019-04-30:
Thanks for looking at it.  I see that you added tessellated sides for the two sides that were missing.  Unfortunately, I want both of those sides to be missing.  What I'm hearing from you is that the only thing I can do is to change my shape so that it's a closed manifold, and give thickness to the entire model where I want it to be printed.  Do I have that right?

Ghostkeeper, 2019-04-30:
Ah. Yeah, you've got that right. Cura is only meant to work with closed manifolds, watertight models. It will try to make something out of non-manifold meshes but it's not expected to work properly.

For one, Cura will not read minds so it cannot determine how thick you wanted your model to be :wink: