Ultimaker/Cura #5678: I placed a model lid on the bed plate then I wanted to place a figure on top of the lid but when I put it into the position I want it in, i.e. 2 or 3 mm into the lid then it immediately drops down onto the bed losing part of the figure, I cant make it stay in the position I want it, please advise. thank you

justme47, 2019-04-30:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

nallath, 2019-04-30:
There are two things that you can do;
1. Group the models together and then use ctrl click to select a model inside the group. Once you did that, you can move it up and it will stay there.
2. Preferences -> Configure Cura -> General -> Disable the "Automatically drop models to the buildplate" preference

closed 2019-04-30 by nallath

justme47, 2019-05-01:
Thank you so much for your time Jaime, the action you describe does work but, saying that, I am sure my graphics card could do with an up grade as it's a bit slow, you solved my problem though and you can tell me to read the instructions, my dad kept telling me for years, lol.