Ultimaker/Cura #5680: Compilation failed

Karaelyn, 2019-04-30:
Hello, 
I try to compile Cura from sources, but at the end I have an error (when I push "python  cura_app.py"). 
my error is  : 

Traceback (most recent call last):
  File "cura_app.py", line 11, in <module>
    from UM.Platform import Platform
  File "/home/engel/dood/Uranium/UM/__init__.py", line 13, in <module>
    from UM.i18n import i18nCatalog
  File "/home/engel/dood/Uranium/UM/i18n.py", line 36
    def __init__(self, name: str = None, language: str = "default") -> None: #pylint: disable=bad-whitespace
                           ^
SyntaxError: invalid syntax

I do everything like the wiki page.
( https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Ubuntu )
All of my dependencies are up to date.
I have surely missed something.
Thanks.
Kara

**Application Version**
Cura 4.0
**Platform**
Ubuntu 18.04

nallath, 2019-04-30:
What version of Python are you using?

Ghostkeeper, 2019-05-01:
The syntax looks correct to me, but it won't be if you're using Python2.

Karaelyn, 2019-05-01:
I use Python 2 effect, is it a problem ?

nallath, 2019-05-02:
Yes. You need python 3.6

closed 2019-05-02 by nallath

Karaelyn, 2019-05-02:
okay, thaanks ! :) 
I have python 3.6 now, but i have another error.
Traceback (most recent call last):
  File "cura_app.py", line 131, in <module>
    import Arcus #@UnusedImport
RuntimeError: the sip module implements API v12.0 to v12.4 but the Arcus module requires API v12.6
thanks a lot team !

Ghostkeeper, 2019-05-03:
It looks to me like you have a version of SIP that supports the SIP API v12.0 through 12.4, but libArcus requires the SIP API to be v12.6. Try swapping out SIP for a newer version.

akshayams94, 2023-01-09:
> okay, thaanks ! :) I have python 3.6 now, but i have another error. Traceback (most recent call last): File "cura_app.py", line 131, in import Arcus #@UnusedImport RuntimeError: the sip module implements API v12.0 to v12.4 but the Arcus module requires API v12.6 thanks a lot team !
You should import arcus package

akshayams94, 2023-01-09:
> I use Python 2 effect, is it a problem ?
Use Python 3