Ultimaker/Cura #5681: Drop down profile menu: too easy to accidently switch profiles when opening drop-down menu

ivanmalvin, 2019-05-01:
**Application Version** Cura 4.0.0

**Platform** Windows 10 x64

**Printer** CR-10S

**Steps to Reproduce** Quickly click on profile drop down menu, then immediately move mouse cursor downward, before the menu appears. (Has to be quite fast.)

**Actual Results** Often it will select whatever profile your cursor was on, and auto-close the drop-down menu right as it appears.

I am guessing this behavior is intended to allow someone to long-click on the drop down menu, keep the mouse button held down, hover over a profile, and release the click to select it. One option would be to require a longer click for the "select on release" functionality to activate

**Expected results** Ideally it would require a click to open the drop down profile menu, and a  **separate** click to select a profile. Instead it seems it selects a profile on click release.

**Additional Information** This just started happening in Cura 4.0.0. When trying to save or update changes I made to a profile, this is happening a lot. When it happens I lose all the changes I had made to the settings. I haven't seen drop down menus work like this in other programs.

Ghostkeeper, 2019-05-01:
I can't reproduce this because the drop-down for quality profiles always appears immediately.

> I am guessing this behavior is intended to allow someone to long-click on the drop down menu, keep the mouse button held down, hover over a profile, and release the click to select it.

As far as I know this is normal for all drop-downs. At least the one in Firefox right above where I'm typing now works like that.

ivanmalvin, 2019-05-03:
Thanks for the reply and testing on your end. I went ahead and made a quick recording of this issue to illustrate exactly what I mean:
[](https://youtu.be/gdNS6iJcRxE
)
Each time in the video I am only single clicking. The issue doesn't happen every time, but I can reproduce it about 50-75% of the time when I try. (And it happens often enough when I'm just using cura normally to be an annoyance.) 

Yes being careful and making sure I don't move the cursor when clicking does fix it, but I'd still consider it to be a bug, or at least something that could be improved.

Ghostkeeper, 2019-05-03:
Ah, I think the difference between our tests is that I had only 6 profiles which didn't cause the dropped down window to go over my cursor.

Ghostkeeper, 2019-05-03:
These drop-downs are created by your operating system, so we don't really have a way to modify them. You could file a feature request with these instructions: https://docs.microsoft.com/en-us/sccm/core/understand/find-help#product-feedback (but don't keep your hopes up).

Normally with any feature that has an option for dragging, like these Windows drop-downs, you need a lenience so that moving by just a few pixels doesn't constitute "dragging" yet because many people move the mouse a little bit when clicking. Cura has this when you try to drag the camera around. You have to move the mouse for a good ~8 pixels while holding the mouse button in order to get the camera to rotate at all.

closed 2019-05-03 by Ghostkeeper

ivanmalvin, 2019-05-03:
I realized part of the problem is I had 20+ profiles, so some of the drop down menu contents actually appeared **on and above** the drop down box, rather than just below as normally expected. Then when that initial mouse click is released, it ends up that your mouse is on top of a profile.

Removing enough profiles to get the contents to only appear **below** the box much fixes the issue from happening unless I'm really flicking the mouse down quickly, on purpose.

Maybe an option would be to limit the height of the drop down profile selection contents (have it scroll when it gets long instead?) Anyway, thanks for looking into it.

Ghostkeeper, 2019-05-06:
I believe Windows will only limit the height of its drop-downs when the drop-down has the full height of the screen. We have no control over that.

We can't disable the entries temporarily either, since the entries will only appear enabled again after refreshing the drop-down. We could make the drop-down empty at first and fill it with entries later, but that will appear weird and buggy.