Ultimaker/Cura #5682: Cura crash when click print via USB

clickphoto, 2019-05-02:
Cura crash when click print via USB.

Win 10 ( 64 ) all drivers updated.
Printer runs on Marlin 1.1.9

nallath, 2019-05-02:
Could you post the logs?

clickphoto, 2019-05-02:
2019-05-02 09:57:59,283 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin ConsoleLogger
2019-05-02 09:57:59,310 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin CuraDrive
2019-05-02 09:57:59,357 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [81]: Found CuraEngine at: C:\Program Files\Ultimaker Cura 4.0\CuraEngine.exe
2019-05-02 09:57:59,365 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin CuraEngineBackend
2019-05-02 09:57:59,384 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin CuraProfileReader
2019-05-02 09:57:59,403 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin CuraProfileWriter
2019-05-02 09:57:59,421 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin OBJReader
2019-05-02 09:57:59,440 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin OBJWriter
2019-05-02 09:57:59,466 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin STLReader
2019-05-02 09:57:59,486 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin STLWriter
2019-05-02 09:57:59,502 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin FileLogger
2019-05-02 09:57:59,527 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin FirmwareUpdateChecker
2019-05-02 09:57:59,552 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin FirmwareUpdater
2019-05-02 09:57:59,575 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin GCodeGzReader
2019-05-02 09:57:59,596 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin GCodeGzWriter
2019-05-02 09:57:59,616 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin GCodeProfileReader
2019-05-02 09:57:59,653 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin GCodeReader
2019-05-02 09:57:59,677 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin GCodeWriter
2019-05-02 09:57:59,701 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin ImageReader
2019-05-02 09:57:59,722 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin LegacyProfileReader
2019-05-02 09:57:59,743 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin LocalContainerProvider
2019-05-02 09:57:59,771 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin LocalFileOutputDevice
2019-05-02 09:57:59,795 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin MachineSettingsAction
2019-05-02 09:57:59,820 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin ModelChecker
2019-05-02 09:57:59,846 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin MonitorStage
2019-05-02 09:57:59,875 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin PerObjectSettingsTool
2019-05-02 09:57:59,900 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin PostProcessingPlugin
2019-05-02 09:57:59,926 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin PrepareStage
2019-05-02 09:57:59,950 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin PreviewStage
2019-05-02 09:57:59,986 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin RemovableDriveOutputDevice
2019-05-02 09:58:00,035 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin SimulationView
2019-05-02 09:58:00,061 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin SliceInfoPlugin
2019-05-02 09:58:00,087 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin SolidView
2019-05-02 09:58:00,114 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin SupportEraser
2019-05-02 09:58:00,149 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.getAccessToken [106]: No auth data to retrieve the access_token from
2019-05-02 09:58:00,156 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin Toolbox
2019-05-02 09:58:00,187 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin CameraTool
2019-05-02 09:58:00,217 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin MirrorTool
2019-05-02 09:58:00,250 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin RotateTool
2019-05-02 09:58:00,281 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin ScaleTool
2019-05-02 09:58:00,339 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin SelectionTool
2019-05-02 09:58:00,374 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin TranslateTool
2019-05-02 09:58:00,421 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin UFPWriter
2019-05-02 09:58:00,453 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin UltimakerMachineActions
2019-05-02 09:58:00,644 - DEBUG - [MainThread] UM3NetworkPrinting.src.Cloud.CloudOutputDeviceManager._onLoginStateChanged [57]: Log in state changed to False
2019-05-02 09:58:00,645 - DEBUG - [MainThread] UM3NetworkPrinting.src.Cloud.CloudOutputDeviceManager._onGetRemoteClustersFinished [80]: Parsed remote clusters to []
2019-05-02 09:58:00,646 - DEBUG - [MainThread] UM3NetworkPrinting.src.Cloud.CloudOutputDeviceManager._onGetRemoteClustersFinished [81]: Removed: 0, added: 0, updates: 0
[... cut ...]
2019-05-02 10:02:58,213 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [417]: Shutting down cura
2019-05-02 10:02:58,322 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [295]: Attempting to kill the engine process
2019-05-02 10:02:58,322 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [301]: Killing engine process
2019-05-02 10:02:58,338 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [304]: Engine process is killed. Received return code 1

clickphoto, 2019-05-02:
[MainThread] UM.Qt.QtApplication.__init__ [76]: Adding QT5 plugin path: C:\Program Files\Ultimaker Cura 4.0\PyQt5\plugins
[MainThread] UM.Qt.QtApplication.__init__ [86]: Adding QT5 plugin path: C:\Program Files\Ultimaker Cura 4.0\PyQt5\plugins
[MainThread] UM.Resources.__initializeStoragePaths [414]: Initializing storage paths
[MainThread] UM.Resources.__initializeStoragePaths [425]: Config storage path is C:\Users\apolon\AppData\Roaming\cura\4.0
[MainThread] UM.Resources.__initializeStoragePaths [433]: Data storage path is C:\Users\apolon\AppData\Roaming\cura\4.0
[MainThread] UM.Resources.__initializeStoragePaths [445]: Cache storage path is C:\Users\apolon\AppData\Local\cura\4.0\cache
[MainThread] UM.PackageManager.__init__ [49]: Found bundled packages JSON file: C:\Program Files\Ultimaker Cura 4.0\resources\bundled_packages\cura.json
[MainThread] UM.PackageManager.__init__ [49]: Found bundled packages JSON file: C:\Program Files\Ultimaker Cura 4.0\resources\bundled_packages\uranium.json
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [113]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
[MainThread] UM.Qt.QtApplication.initialize [138]: Detected most suitable OpenGL context version: 4.1 Core profile
[MainThread] UM.Qt.QtApplication.initialize [144]: Initializing job queue ...
[MainThread] UM.Qt.QtApplication.initialize [148]: Initializing version upgrade manager ...
[MainThread] UM.Settings.ContainerRegistry.addContainer [386]: Container [empty_definition_changes] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [386]: Container [empty_variant] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [386]: Container [empty_material] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [386]: Container [empty_quality] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [386]: Container [empty_quality_changes] added.
[MainThread] UM.PackageManager._loadManagementData [83]: Loaded bundled packages data from C:\Program Files\Ultimaker Cura 4.0\resources\bundled_packages\cura.json
[MainThread] UM.PackageManager._loadManagementData [83]: Loaded bundled packages data from C:\Program Files\Ultimaker Cura 4.0\resources\bundled_packages\uranium.json
[MainThread] UM.PackageManager._loadManagementData [100]: Loaded user packages management file from C:\Users\apolon\AppData\Roaming\cura\4.0\packages.json
[MainThread] UM.PackageManager._saveManagementData [168]: Package management file C:\Users\apolon\AppData\Roaming\cura\4.0\packages.json was saved
[MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [90]: Loading plugin configuration file 'C:\Users\apolon\AppData\Roaming\cura\4.0\plugins.json'
[MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin 3MFReader
[MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin 3MFWriter
[MainThread] UM.PluginRegistry.loadPlugin [436]: Loaded plugin ChangeLogPlugin

clickphoto, 2019-05-02:
[cura.log](https://github.com/Ultimaker/Cura/files/3137176/cura.log)

clickphoto, 2019-05-02:
[stderr.log](https://github.com/Ultimaker/Cura/files/3137178/stderr.log)
[stdout.log](https://github.com/Ultimaker/Cura/files/3137179/stdout.log)

nallath, 2019-05-02:
I don't see any info regarding a crash, but you do seem to lack permissions to open the COM6 port.

clickphoto, 2019-05-02:
oops.... sorry... I was just clean the logs before.... Please check this files.
[stderr.log](https://github.com/Ultimaker/Cura/files/3137286/stderr.log)
[cura.log](https://github.com/Ultimaker/Cura/files/3137287/cura.log)

clickphoto, 2019-05-02:
stdout  file is empty

clickphoto, 2019-05-03:
I just find out it crash also when save file....not only on print on usb.

[stderr.log](https://github.com/Ultimaker/Cura/files/3140644/stderr.log)
[cura.log](https://github.com/Ultimaker/Cura/files/3140645/cura.log)

diegopradogesto, 2019-05-03:
To me it looks very similar to this issue that we had back in 3.4: https://github.com/Ultimaker/Cura/issues/4016

Could you check the workarounds that we posted there?

Ghostkeeper, 2019-05-03:
Which version of Cura are you using?

clickphoto, 2019-05-03:
Latest version

Ghostkeeper, 2019-05-03:
Could you go into the preferences and disable the "Send (anonymous) print information" preference there, and then try again? If it doesn't crash then, that confirms that the bug in #4016 is back.

clickphoto, 2019-05-03:
Thanks for the respond... IS WORKING.

I can confirm that print on usb and save gcode after Send (anonymous) print information disabled

clickphoto, 2019-05-03:
Please let me know if you need any log.

Thanks again all of you for the help.

GregValiant, 2024-11-15:
Is this still an issue with current Cura versions (5.8.0 and up).  Can this be closed?

closed 2024-12-10 by github-actions[bot]