Ultimaker/Cura #5683: API sip version Runtime Error

Karaelyn, 2019-05-02:
Hello team,
Sorry to ask you again.
I have an error when doing : python3 cura_app.py
the system return me : 

Original exception was:
Traceback (most recent call last):
  File "cura_app.py", line 131, in <module>
    import Arcus #@UnusedImport
RuntimeError: the sip module implements API v12.0 to v12.4 but the Arcus module requires API v12.6

I look on another old topic ( #704   18 Apr 2016 ) but i don't have Cura installed before so the solution doesn't work. :/
I have followed exactly all steps of the tuto.
I precise that my system is completely new, just reinstalled.

Thanks a lot for your greatfull job.
Kara

**Platform**
Ubuntu 18.04
**Steps to Reproduce**
https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Ubuntu#sip

ZaidBarghouthi, 2019-05-03:
Same problem. Although this will be my first time running from source.

LipuFei, 2019-05-03:
Hi @Karaelyn , could you run the following commands to check which SIP version your python3 by default imports? Compare with the SIP version you have installed by following the steps, see if they match.
```
> python3
Python 3.7.1 (v3.7.1:260ec2c36a, Oct 20 2018, 14:57:15) [MSC v.1915 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import sip
>>> sip.SIP_VERSION_STR
'4.19.8'
>>> sip.__file__
'C:\\workspace\\installations\\python\\3.7.1\\lib\\site-packages\\sip.pyd'
>>>
```

Karaelyn, 2019-05-03:
hi @LipuFei , i compare, and have exactly the same version as you & tuto :/
python3
Python 3.6.7 (default, Oct 22 2018, 11:32:17) 
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sip
>>> sip.SIP_VERSION_STR
'4.19.8'
>>> sip.__file__
'/home/kara/.local/lib/python3.6/site-packages/sip.so'
>>> 
same version so ....

ZaidBarghouthi, 2019-05-03:
same here..
`4.19.8`
`/usr/local/lib/python3.6/dist-packages/sip.so`

LipuFei, 2019-05-03:
@Karaelyn @ZaidBarghouthi   Could you for `libArcus`, if you run `cmake`, which SIP does it find? For example, this is what I get:
```
-- Found Protobuf: C:\workspace\installations\protobuf\3.4.1\msvc2017\lib\libprotobuf.lib (found suitable version "3.4.0", minimum required is "3.0.0")
-- Found Python3: C:/workspace/installations/python/3.7.1/python3.exe (found suitable version "3.7.1", minimum required is "3.4") found components:  Interpreter Development
-- Found SIP: C:/workspace/installations/python/3.7.1/sip.exe (found version "4.19.8")
```

ZaidBarghouthi, 2019-05-03:
```
-- Found Protobuf: /usr/local/lib/libprotobuf.a;-lpthread (found suitable version "3.7.1", minimum required is "3.0.0") 
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.7", minimum required is "3.4") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable version "3.6.7", minimum required is "3.4") 
-- Found SIP: /usr/bin/sip (found version "4.19.8") 
```

Ghostkeeper, 2019-05-03:
Since you are on Linux, I think the chance is very big that you have multiple versions of SIP. PyQt5 comes with its own SIP which should be used only for PyQt (which is probably the one that requires 12.0-12.4). This version of SIP can be different from the version in /usr/bin/sip.

LipuFei, 2019-05-03:
@ZaidBarghouthi That SIP executable path doesn't look right. Your previous result says that the sip dynamic library is in `/usr/local/lib/python3.6/dist-packages/sip.so`, so your sip module should be installed in `/usr/local`, but cmake finds `/usr/bin/sip` instead of `/usr/local/bin/sip`. Could you try this?
```
cmake -DCMAKE_PREFIX_PATH=/usr/local <any-extra-arguments-that-you-need>
```

ZaidBarghouthi, 2019-05-03:
I'm not sure how to run this

```
labarg@labarg-dell:~/projects/libArcus/build$ cmake .. -DCMAKE_PREFIX_PATH=/usr/local 
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Protobuf: /usr/local/lib/libprotobuf.a;-lpthread (found suitable version "3.7.1", minimum required is "3.0.0") 
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.7", minimum required is "3.4") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable version "3.6.7", minimum required is "3.4") 
-- Found SIP: /usr/bin/sip (found version "4.19.8") 
-- Configuring done
-- Generating done
```

ZaidBarghouthi, 2019-05-03:
> Since you are on Linux, I think the chance is very big that you have multiple versions of SIP. PyQt5 comes with its own SIP which should be used only for PyQt (which is probably the one that requires 12.0-12.4). This version of SIP can be different from the version in /usr/bin/sip.

@Ghostkeeper Any suggestions?

LipuFei, 2019-05-03:
Do you have SIP in /usr/local/bin? Could you do a `find / -name 'sip' -type f` to show all possible sip executables that you have?

ZaidBarghouthi, 2019-05-03:
No, only
`/usr/bin/sip`
`home/labarg/projects/sip/sipgin/sip`

@Ghostkeeper I uninstalled PyQt5 and got the same error. I guess the sip error occurs before importing PyQt5

ZaidBarghouthi, 2019-05-04:
@Karaelyn did you manage to run it?

Karaelyn, 2019-05-06:
hello team, sorry for the delay.
The libArcus's cmake send : 
-- Found Protobuf: /usr/local/lib/libprotobuf.a;-lpthread (found suitable version "3.7.1", minimum required is "3.0.0") 
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.7", minimum required is "3.4") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable version "3.6.7", minimum required is "3.4") 
-- Found SIP: /usr/bin/sip (found version "4.19.8")

Karaelyn, 2019-05-06:
and : 
$ sudo find / -name 'sip' -type f
find: ‘/run/user/1000/gvfs’: Permission non accordée
/usr/bin/sip
thanks for your help team

sissi-luaty, 2019-05-16:
Hi, I had the same problem in Linux kubuntu 18.04 (one with a fresh install, and another full of stuff). Today I found the solution for the problem on my pc, which was similar to the one referred here.

The problem occurs when the version of sip is wrong (lower than required), or mixed (two versions, with one of them lower than required).

Type, in a python shell:
`import sip; print(sip, sip.SIP_VERSION_STR)`

and in a system shell:
`sip -V`

Now, if you are having this issue, the versions do not match or are very small. After the following they will match.

-- in order to solve this problem, I did a remove of the system's sip old version:
`sudo apt-get purge python3-sip`
-- and then I reinstalled the latest release of sip-4.19.17.tar.gz from: "https://www.riverbankcomputing.com/software/sip/download", doing the commands from within the extracted folder of sip:
```
python3.6 configure.py
make
sudo make install
```

Hope that helps!

Ghostkeeper, 2019-05-17:
That's in fact what I also did in my build environment, sissi-luaty! Sip in general is kind of hard to compile with, I think, so I've always used my own build of it to make sure that I've used the same version everywhere.

closed 2019-05-17 by Ghostkeeper