Ultimaker/Cura #5685: [4.0.0] Cura adding top and bottom layers where none exist and then adding infill.

TimCraig, 2019-05-03:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

[Cylinder.zip](https://github.com/Ultimaker/Cura/files/3140416/Cylinder.zip)

smartavionics, 2019-05-03:
Hello @TimCraig , in your profile, the Remove All Holes setting was enabled (oops!) Cura was just doing your bidding. Hope this helps.

[image]

TimCraig, 2019-05-03:
Was "Remove All Holes" enabled by default in 4.0.0.  It was not not visible in my profile in either 4.0.0 or 3.5.1.   In 3.5.1 it was not enabled.  I didn't enable it in either version.

smartavionics, 2019-05-03:
Unlikely that it was enabled by default as that would cause mass hysteria.

Ghostkeeper, 2019-05-03:
The "Remove All Holes" setting was not enabled by default in 4.0 (nor was it ever):

https://github.com/Ultimaker/Cura/blob/59b3046c0f04a357593f18434a286832fd072a89/resources/definitions/fdmprinter.def.json#L5379-L5386

The setting is also not overwritten by Creality Ender 3 or the one in that project file, Creality CR10 (which was renamed to Creality Ender-3).

In your project file it shows that it was enabled by the user, not in any of the profiles. I think you must've clicked it accidentally.

closed 2019-05-03 by Ghostkeeper