Ultimaker/Cura #5688: disappearing icons

PsyonicOne, 2019-05-03:
**Application Version**
4.0.0

**Platform**
Linux running Fedora 29 with Gnome 3.30.2
Nvidia GeForce RTX 2070 Driver version 418.56

**Printer**
Ender 3

**Steps to Reproduce**
It seems to happen when Gnome puts the system to sleep automatically after no activity, then after waking back up, I can not reproduce it after simply blanking the screen, it's always after an auto power-off sleep.

**Actual Results**
Side icons (move, scale, etc.), the ticks in the boxes in the settings, the top bar menu (File, Edit, View, etc.) cannot be seen. See image below, yellow circles.
Also, the blue bar with the Cura logo cannot be seen.

**Expected results**
Above should be still visible.

**Additional Information**
I haven't tried this with any other Fedora/Gnome versions but this did not happen with past versions using my current setup, and no other program seems to be having issues.

If I run my mouse over the menu bar the File, Edit, View, etc. does return, but none of the other features return in the same way.

The only way to correct the problem is to close Cura and reopen, but if the system goes to sleep again the problem returns.

(P.S. Thank you for your amazing software!!!)

[image]

Ghostkeeper, 2019-05-03:
Doesn't reproduce on Ubuntu 19.04, which uses Gnome as well. @nallath Which Fedora version were you running?

> did not happen with past versions using my current setup

Did you mean with past Cura versions?

PsyonicOne, 2019-05-04:
Fedora 29 with Gnome 3.30.2

Yep, I did mean past versions of Cura (3.6.x) on this rig with same Fedora and Gnome (I'll update the issue comments)

PsyonicOne, 2019-06-03:
This is still occurring in 4.0.1

Ghostkeeper, 2019-06-03:
I suppose you meant 4.1.0.

We are still unable to reproduce this bug on any of our computers (including Fedora). We therefore haven't made an effort to change anything.

Ghostkeeper, 2019-06-03:
Maybe if more people can reproduce this issue we'll be able to figure out a pattern as to what computers have this problem and what computers don't.

PsyonicOne, 2019-06-05:
Thanks @Ghostkeeper 
It might have something to do with the Nvidia driver. I'll update it when I can and report.

Woobieman29, 2019-10-11:
I can reproduce this issue on my workstation using PopOS v19.04 which is similar to Ubuntu. I will post screenshots and some basic info on my machine, happy to provide anything needed to troubleshoot. This happens to me the same way as it seems to with OP - only when the machine goes to sleep by itself on the activity timer. This has been happening for a few months at least... not sure when it started. 

PopOS 19.04 / AMD64
Cura v4.2.1
NVIDIA GTX750Ti graphics card with NVIDIA 430.34 drivers 
[image]

Woobieman29, 2019-10-11:
Note one other difference on my screen compared to OP is the groovy rainbow effect at the top of the window that obscures some of the menu text, the Cura logo, etc. 

Also, thank you for providing Cura! :)

Woobieman29, 2019-10-11:
I have some entries from syslog that might be helpful, attached as a text file. Shows cura.desktop activity as the machine was going to sleep, when it was waking up, and a couple messages that have happened since. 
[CuraIssueLog.txt](https://github.com/Ultimaker/Cura/files/3719056/CuraIssueLog.txt)

Ghostkeeper, 2019-10-14:
From the log:

> Oct 11 08:29:49 toes cura.desktop[1642]: QXcbConnection: XCB error: 148 (Unknown), sequence: 3217, resource id: 0, major code: 140 (Unknown), minor code: 20

This seems to have something to do with your `/etc/x11/xorg.conf` file being wrong, according to [this StackOverflow answer](https://stackoverflow.com/a/45341303).

@ninovanhooff told me that this issue with corrupt textures happens with NVidia GPUs after resuming from sleep. Chrome apparently has it too. I think then that there is nothing that Cura can really do about it. We should be able to expect textures to remain loaded in the GPU...

See also https://bugreports.qt.io/browse/QTBUG-57845. This is upstream and somewhat out of our hands, so I'll close this issue.

closed 2019-10-14 by Ghostkeeper