Ultimaker/Cura #5689: Gaps

henkie1, 2019-05-03:
Hello
Design from tinkercad,
When i slice it in Cura in the preview there are 2 gaps in the line. 

[image]

smartavionics, 2019-05-03:
The wall must be too thin. Try enabling the Print Thin Walls setting or reducing the wall line width.

Ghostkeeper, 2019-05-06:
Please supply some reproduce steps or a project file. I agree that it's probably that the model is thinner than the line width. Or else it could be coasting (but then I would expect only one hole).

henkie1, 2019-05-06:
Well, I tried a lot but nothing helps.
Attached the stl file. Perhaps sombody gives it a try.

[dalia 1,5.stl.txt](https://github.com/Ultimaker/Cura/files/3148958/dalia.1.5.stl.txt)

henkie1, 2019-05-06:
I think I have got it. Option "print thin walls" on.  It seems its ok now.
I print wil print it this evening and let you know.
[image]

Ghostkeeper, 2019-05-07:
The thinnest part of your model (the tip of the leaf left of the most negative Y coordinate) is 0.347mm thick. If you set the line width to 0.347mm or lower, you'll find that all of the walls print continuously without gaps. Print Thin Walls is an easier alternative (but will result in some extra travel moves).

closed 2019-05-07 by Ghostkeeper