Ultimaker/Cura #5690: [4.X GIT] Unnecessary Nozzle Move Every Layer

Enverex, 2019-05-03:
I've never seen this before and no-one I've asked has been able to explain what is happening here. Basically, at the start of every layer, the nozzle will move to the front of the print,  move back and forth a bit and then carry on with the print. It's hard to explain so here's a video of it happening - https://www.youtube.com/watch?v=sRfVzFExcMQ

This "dance" is even visible in sliced view itself (the purple line closest to the camera in the bottom left). 
[image]

It doesn't extrude when it does this, it purely seems to be a travel move, but with no obvious purpose. For reference, I've attached the 3MF file of the model/setup.
[Ocarina.zip](https://github.com/Ultimaker/Cura/files/3143692/Ocarina.zip)

smartavionics, 2019-05-04:
Sorry, I can't investigate this as the project file won't load. Can you create another user a standard printer type? Thanks.

[image]

smartavionics, 2019-05-04:
Actually, just attaching the gcode would probably be good enough, thanks.

Enverex, 2019-05-04:
Here you go. 
[Ocarina.zip](https://github.com/Ultimaker/Cura/files/3144314/Ocarina.zip)

smartavionics, 2019-05-04:
OK, I discovered what the problem was, you had the Wipe Nozzle Between Layers feature enabled.

[image]

Enverex, 2019-05-04:
Is there anything else that would enable this? As it's not something I would have turned on (due to not having a multi-head setup). Also, what does it think it's wiping it on?

smartavionics, 2019-05-04:
> Is there anything else that would enable this? As it's not something I would have turned on (due to not having a multi-head setup).

No, there is nothing else that would enable this.

> Also, what does it think it's wiping it on?

Good question, I don't know. I guess the creator of this feature had a brush mounted somehow.

smartavionics, 2019-05-04:
FYI https://github.com/Ultimaker/Cura/pull/5356

Enverex, 2019-05-04:
Thanks, I'll chalk it up to an unnoticed misclick at some point then.

closed 2019-05-04 by Enverex

Ghostkeeper, 2019-05-06:
> Also, what does it think it's wiping it on?

Yeah, this is intended for printers with a wiping brush.