Ultimaker/Cura #5691: Curved sharp corners result in tiny deposits and excessive travel that run nozzle dry

larsch, 2019-05-04:
**Application Version**: 4.0

**Platform**: Windows & Linux

**Printer**: Custom FDM (Ultimaker Clone)

**Steps to Reproduce**: Slice model with walls/shells meeting at sharp curved angles. Example attached (problem in layers 9-62 at 0.1 mm layer height.

[tiny_blob_deposits_in_corners.zip](https://github.com/Ultimaker/Cura/files/3144098/tiny_blob_deposits_in_corners.zip)

**Actual Results**: Cura generates gcode that, at the very end of each layer, performs extreme amounts of travel moves to deposit tinyl blobs in every corner where two shells meet. Nozzle runs dry during all of these travel moves and the first 5 cm of deposits of infill on next layer is non-existent.

[image]

[image]

**Expected results**: The blobs deposited in the corners should not really even be a part of the necessary geometry. They should also really just be eliminated as it is pointless to try and deposit such small amounts. An option to eliminate short extrusions (minimum extrusion length) would help, but I don't really think that the geometry warrants those blobs in the first place.

**Additional Information**

smartavionics, 2019-05-04:
Hello @larsch , please save the project file (File -> Save), zip the .3mf file and attach to this thread. That way I can see the settings you are using. Thanks.

smartavionics, 2019-05-04:
Hello again, I worked out that those little areas in the corners are where it is filling gaps between walls so if you turn off that option they will go away. To avoid dribble on long moves, I recommend that you set the max comb distance with no retract to something like 20. Hope this helps.

larsch, 2019-05-04:
Thanks for the tips. Removing gap fill does indeed work around it. 

Max comb distance in this case would need to be very low (all travel moves are < 20 mm), but good tip for other cases.

[tiny_blob_deposits_in_corners.3mf.zip](https://github.com/Ultimaker/Cura/files/3144201/tiny_blob_deposits_in_corners.3mf.zip)

smartavionics, 2019-05-04:
Thanks for the project file.

Unless it causes any problems (which would warrant an issue being filed), it's generally advantageous to enable the Optimize Wall Printing Order setting as this should reduce travels. I also think that using the no-skin combing mode is best as that avoids travels over skins which tend to leave ugly lines.

Ghostkeeper, 2019-05-06:
This seems to be behaving as expected, I'd say. This gap filling is meant to fill the gap where the line width is too fat to reach, which is why it's only happening in sharp corners. It needs to happen all in one go since it interrupts the flow quite a lot. Otherwise you'd have 10 interruptions instead of just 1. However in your case with the required detours and the narrow gaps this introduces some long travel moves that are better off retracted.

The default settings will not work perfectly for all models. I'm sorry. But that's why they are settings, after all!

closed 2019-05-06 by Ghostkeeper

larsch, 2019-05-07:
But there are no gaps. Just two shells meeting at a sharp but curved angle. That should will not make a gap, and shouldn't require gap filling. S3D never generates such nonsense.

Ghostkeeper, 2019-05-07:
Whenever two shells meet at a sharp angle and another line is meant to fill the inside corner, there will be a gap because the line that fills the inside corner cannot reach all the way into the corner. Here's an illustration of how it is modelled, from our [documentation](https://github.com/Ultimaker/CuraEngine/blob/master/docs/generating_paths.md#generating-walls):

https://github.com/Ultimaker/CuraEngine/blob/master/docs/assets/gaps_in_walls.svg

If S3D never generates such nonsense, your prints will be weaker when printing with S3D (in many prints).

The difference here is, I guess, that you don't necessarily have a line that fills the inside corner if your infill is sparse. That's what makes this look weird. We're considering not filling gaps within the innermost border if infill is less than a certain percentage (like 50% or so). That would save some travel moves in your case. But that project has been deferred for some time.

larsch, 2019-05-07:
The "blobs" in my screenshot above are clearly on the inside of the inner shell (and in seemingly every single corner, regardless of how the infill interacts). The triangles in gaps_in_walls.svg are between the shells. I still don't understand why there needs to be a blob on the inside of the inner shell perimeter.

[image]