Ultimaker/Cura #5692: [4.0.0] Login - Unsupported method ('HEAD')

MSergW, 2019-05-04:
**Application Version**
4.0.0

**Platform**
Debian

**Printer**
ender-3

**Actual Results**
Login - not work

When I press the button - login - nothing happens, such information appears in the logs

2019-05-04 11:15:45,649 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.startAuthorizationFlow [139]: Starting new OAuth2 flow...
2019-05-04 11:15:45,665 - DEBUG - [MainThread] cura.OAuth2.LocalAuthorizationServer.start [45]: Starting local web server to handle authorization callback on port 32118
127.0.0.1 - - [04/May/2019 11:15:45] code 501, message Unsupported method ('HEAD')
127.0.0.1 - - [04/May/2019 11:15:45] "HEAD /callback?error=invalid_request&error_description=methodNotAllowed%3A+The+requested+HTTP+method+is+not+allowed.&state=%28.Y.%29 HTTP/1.1" 501 -

Ghostkeeper, 2019-05-06:
:microphone: :notes:

Ghostkeeper, 2019-05-06:
It seems that the log-in server has responded that you're not allowed to use HTTP. And then that response in itself is not understood by Cura.

Does your browser use HTTPS when you're logging in? Does it say "https://" in the URL on the top of the screen?

I'm unable to reproduce this issue so far.

jellespijker, 2019-06-10:
Hi @Ghostkeeper I have a similar issue on my personal Manjaro machine

When I press sign-in the following message is shown

```
gio: https://account.ultimaker.com/authorize?client_id=um----------------------------ultimaker_cura&code_challenge=GDadrwKdt-gvRv2tWsjHW9YLu-pxsk582QKjNkqTNATO4vx759EaVja_cQYLYX6Bl1A_aL_22w3QGpa_qNVg9A%3D%3D&redirect_uri=http%3A%2F%2Flocalhost%3A32118%2Fcallback&code_challenge_method=S512&state=%28.Y.%29&response_type=code&scope=account.user.read+drive.backup.read+drive.backup.write+packages.download+packages.rating.read+packages.rating.write+connect.cluster.read+connect.cluster.write+cura.printjob.read+cura.printjob.write+cura.mesh.read+cura.mesh.write: HTTP Error: Unsupported method ('HEAD')
```

Linux jakatu 5.1.4-1-MANJARO #1 SMP PREEMPT Wed May 22 17:56:21 UTC 2019 x86_64 GNU/Linux

Running: Cura-4.1.0

If you like I could take my laptop to work. Just mail to j.spijker@ultimaker.com

Ghostkeeper, 2019-06-10:
> If you like I could take my laptop to work.

That would be useful, yeah!

Ghostkeeper, 2019-06-20:
A workaround would be to go into your log file (https://github.com/Ultimaker/Cura#logging-issues) find the error message and copy the link there into your browser while Cura is running. It should log you in.

We're looking into what causes this. Something with xdg maybe and your default browser not being configured correctly?

pkuiper-ultimaker, 2019-06-21:
Issue is being investigated by our Cloud team

marijnd, 2019-06-21:
I can reproduce this (or at least something very similar) in Manjaro:

- Install Cura from the package manager
- Start the browser and log into you Ultimaker account via https://account.ultimaker.com
- Close the browser
- Start Cura
- Press Sign In -> Sign In
- Error in the browser 'Unable to connect' with url 'http://localhost:32118/callback?code=VySIoVzjStkAQL32HTJyo2zq637WO5M7&state=%28.Y.%29'

The error does not occur when:
- Start Cura from the Ultimaker Cura 4.0.0 AppImage
- Press Sign In -> Sign In

Reporter could test whether the error also occurs with the cura appimage.

pkuiper-ultimaker, 2019-06-21:
OK, the issue reported by marijnd seems to be a different issue. It has to do with the package manager which we do not support. The AppImage is supported and works.
 Marijn and I also tried the Cura 4.1 AppImage version on a Manjaro machine with Firefox and this also work OK. What kind of browser are you using on Manjaro?

In the mean time I'll investigate if we can let the local webserver respond to a HEAD request.

pkuiper-ultimaker, 2019-06-24:
Developers see: CURA-6609

jellespijker, 2019-06-24:
@pkuiper-ultimaker I use chrome on my personal computer Manjaro computer

jellespijker, 2019-06-25:
Additional information:

before I tried the appimage I installed and uninstalled Cura 4.0 from the AUR:
But I purged the three caches afterwards.
```
prepare(){
  cd Cura-${pkgver}
  sed -i 's,/dist-packages,.${PYTHON_VERSION_MINOR}/site-packages,g' CMakeLists.txt
}

build(){
  cd Cura-${pkgver}
  mkdir -p build
  cd build
  cmake .. \
    -DCMAKE_INSTALL_PREFIX=/usr \
    -DCMAKE_BUILD_TYPE=Release \
    -DURANIUM_SCRIPTS_DIR=/usr/share/uranium/scripts \
    -DCURA_VERSION=$pkgver \
    -DCURA_SDK_VERSION=6.0.0 -DCURA_CLOUD_API_VERSION=1

  make
}

package(){
  cd Cura-${pkgver}/build
  make DESTDIR="${pkgdir}" install
  
  # make sure cura can find uranium plugins:
  ln -s /usr/lib/uranium/plugins/* "${pkgdir}/usr/lib/cura/plugins/."

  # don't ever send any user or print info through the internet to Ultimaker
  rm -rf "${pkgdir}/usr/lib/cura/plugins/SliceInfoPlugin"
}
```

LipuFei, 2019-08-27:
Fixed in 9f110368d