Ultimaker/Cura #5696: [4.0.0] Support Line Pattern results in zigzag

pvd, 2019-05-05:
**Application Version**
4.0.0

**Platform**
macOS Mojave (10.14.4)

**Printer**
ANet A6 with Marlin 1.1.8
[CFFFP_DP_Torso.3mf.zip](https://github.com/Ultimaker/Cura/files/3145271/CFFFP_DP_Torso.3mf.zip)

**Steps to Reproduce**
1. Load a stl file, I've tried various stl file and the problem is the same of all of them.
2. Select a layer height of 0.06 mm
3. Enable supports and select the line pattern
4. Slice; the preview will show the line support pattern
5. Copy G-code to sdcard and print

**Actual Results**
The printer is making a zigzag support pattern instead of the expected line pattern depicted in the Cura preview.

**Expected results**
The support pattern shown in the preview to match the generated/printed support pattern.

**Additional Information**

[image]

[image]

smartavionics, 2019-05-05:
Just guessing here, perhaps it really is using lines but dribbling between the line ends which makes it look like zig-zag?

pvd, 2019-05-05:
> Just guessing here, perhaps it really is using lines but dribbling between the line ends which makes it look like zig-zag?

I don't thinks so, the print just finished and it really looks like a zigzag pattern.
[image]

I'm now printing with the zigzag pattern and the resulting pattern is exactly the same.

smartavionics, 2019-05-05:
In that case, you probably have the Connect Support Lines setting enabled.

[image]

pvd, 2019-05-05:
> In that case, you probably have the Connect Support Lines setting enabled.
> 
> [image]

That option is hidden when de pattern is set to lines. 
What puzzles me is that it preview is different from what is actually executed by my printer.
Later I will try the same file with Cura 3.6.0

Ghostkeeper, 2019-05-06:
One thing you could try is to save the g-code to file on your computer and load it back in.

When printing these settings it looks normal on my printer.

[image]