Ultimaker/Cura #5697: [4.1] Wrong text inserted into printer name label

thopiekar, 2019-05-05:
**Application Version**
All built from 4.1 branches

**Platform**
Doesn't matter.

**Printer**
Doesn't matter.

**Steps to Reproduce**
Choose "Monitor" tab

**Actual Results**
Label shows "Extruder 1"

**Expected results**
Label should show the printers name

**Additional Information**

[image]

[image]

[image]

Ghostkeeper, 2019-05-06:
I could reproduce this, but only when switching to a printer that wasn't created during the same session. So these were my reproduction steps:
1. Add two printers.
2. Restart Cura.
3. Switch to the Monitor page.
4. Switch printers.

thopiekar, 2019-05-06:
Ok, had both printers added already.

>

thopiekar, 2019-05-15:
Has the problem been solved for the beta? Or is it going to be resolved before the final release?
Personally, I would find it wise to care about getting that fixed.

I think it is an annoying one.

Ghostkeeper, 2019-05-17:
Ah yeah we fixed this for 4.1. It was in fact also happening in the main printer selection drop-down in the prepare stage.

closed 2019-05-17 by Ghostkeeper

thopiekar, 2019-05-17:
Great, thanks! 👍