Ultimaker/Cura #5699: 3.6 and 4.0 crash

Donner-Party, 2019-05-05:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

Donner-Party, 2019-05-05:
3.6.0 was working fine for me the last time I used it on 4/16/2019.
I tried to use it today and it would not show the UI.
I uninstalled 3.6.0 and installed 4.0.
4.0 starts, shows the "changelog" window for a second and then disappears.

The only things that have changed since the last time it worked are regular windows updates and I had to update the display driver because the backlight dimming had stopped working.

Files attached.
[cura.log](https://github.com/Ultimaker/Cura/files/3145893/cura.log)
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/3145894/DxDiag.txt)

Ghostkeeper, 2019-05-06:
I think this is a duplicate of https://github.com/Ultimaker/Cura/issues/5651.

If you delete the file `C:\Users\<Donner-Party>\AppData\Roaming\cura\4.0\cura.cfg`, does it work properly again then?
Deleting that file resets your preferences and causes the add printer screen to appear again, but it shouldn't delete your profiles or added printers.

Donner-Party, 2019-05-06:
Thanks, I'll try it when I get home.

I don't have multiple monitors like #5651.

Donner-Party, 2019-05-07:
Tried deleting the specified file. 
Cura starts and displays this dialog box and then disappears:
  [image]

Ghostkeeper, 2019-05-07:
Ah, so no cigar and not even close.

I have no real ideas on how to begin debugging this. It's a blocking issue for some people, but if we're unable to reproduce it here and the logs say nothing, we can only guess in the dark.

Ghostkeeper, 2019-05-07:
Maybe if we get more reports like this we can figure out what the commonalities between those users are.

Donner-Party, 2019-05-18:
[image]

When I disable Intel UHD Graphics 620, I found that Cura 4.0 runs normally, but I lose the ability to control screen resolution and screen brightness on my laptop.

Ghostkeeper, 2019-05-20:
That's a nice find. I wonder what weirdness that driver is doing that prevents Qt from drawing anything.

sedwards2009, 2019-05-20:
I had a strange problem the other day with Cura and OpenGL on my work laptop in combination with DisplayLink monitors across USB3. It looks like booting with the USB3 connected and the (laptop) screen off sets up a default OpenGL "environment" which fails to initialise properly. Booting and logging in with the laptop screen and *then* switching to monitors over USB3 did work.

Donner-Party, 2019-06-06:
The problem continues with 4.1.
If I disable Intel UHD Graphics 620, Cura runs normally.
If I enable 620 while Cura is running it continues running, but I can't open a new instance of Cura. In the previously opened instance of Cura, I can move the workplane and model around and the slicer works, but none of the menu items respond.

Donner-Party, 2020-01-10:
I notice if I re-enable UHD 620 after Cura is started, it continues to run normally.

Is there a command line switch to turn off the splash screens when Cura starts?

Ghostkeeper, 2020-01-14:
I don't think the splash screen is causing the actual problem. It's about as simple as it gets. Maybe it's got more to do with creating a new window and canvas?

Donner-Party, 2020-02-19:
This Intel update [https://downloadcenter.intel.com/download/29335/Intel-Graphics-Windows-10-DCH-Drivers?product=126789](https://downloadcenter.intel.com/download/29335/Intel-Graphics-Windows-10-DCH-Drivers?product=126789) seems to have fixed my problems with Cura, Fusion360, and DWG TrueView.

closed 2020-02-25 by Ghostkeeper