Ultimaker/Cura #5700: [4.0.0] Ironing even where things are printed on top

nh2, 2019-05-05:
**Application Version**
4.0.0; the screenshot is from 3.6, but the same problem arises in 4.0.0 (I've just tried it).

**Platform**
Ubuntu 16.04

**Printer**
Creality Ender 3

**Steps to Reproduce**
Make some flat surface, like a cube, and put some font on top. I used OpenScad for it.

Enable `Ironing`.

**Actual Results**
The surface behind the font is very ugly, because it is not "ironed through" in one go.

See the screenshot of the layer that is ironed:

[image]

In the places where the font sits on top, the ironing doesn't follow the normal pattern (e.g. inside the lowercase `b`).

**Expected results**
I'd like to achieve that the full layer is ironed, so that it has an equally nice finish everywhere, and that the font is simply printed on top.

If this isn't possible with Cura, currently, I'd like to feature-request it.

nh2, 2019-05-05:
One layer higher, I have this (just for better understanding):

[image]

Ghostkeeper, 2019-05-06:
Indeed the ironing routine does not do this. Something similar can be achieved with the Top Surface Skin Layers setting though, but probably not exactly as you want it.

So as I understand it, your feature request is that ironing should iron the entire layer completely if anything on that layer is top surface?

So if you were to use it on a shape, like, a pyramid, the entire pyramid would become ironed.

nh2, 2019-05-06:
> So as I understand it, your feature request is that ironing should iron the entire layer completely if anything on that layer is top surface?
> 
> So if you were to use it on a shape, like, a pyramid, the entire pyramid would become ironed.

Yes.

Indeed there are some adversarial shapes where you wouldn't want to use this setting.

But I imagine there can be simple/good selectors for this. For example, for the case where you want to print font on top of an otherwise flat top surface, sensible settings might be:

> **Entire-layer ironing threshold (%)**
> Iron through entire layer if more than N% of the layer would be ironed.

or

> **Iron-through-patch threshold (%)**
> Iron through (instead of around) any contiguous surface patch that is smaller than N% of the layer.

For the latter, in the case of my `b` letter in the first screenshot, let's say that letter takes < 10% of the layer surface; then you could set `Iron-through-patch threshold = 10%` and the patch under the would be ironed through, making for

Ghostkeeper, 2019-05-06:
Oh, well perhaps it would make more sense to do a closing operation on the ironed area then instead.

nh2, 2019-05-06:
@Ghostkeeper I suspect that would also do.

diegopradogesto, 2019-06-20:
There is a FR in our backlog with an internal ticket number CURA-6512. We will inform you when this is picked up. However, considering the work we have planned ahead, this is not going to happen soon (or ever). We'll keep you up to date.

GregValiant, 2024-11-24:
There have been some changes over the years in the ironing function.  A lot more settings are involved.
Can this be closed?

nh2, 2024-11-26:
I haven't gotten the chance yet to test the same model in newer CURA. I will try to do it as soon as my distribution has a current version packaged.